MCUdude/MightyCore #4: Boards Manager installation support

per1234, 2015-12-22:
If you're interested in adding Boards Manager support I'm willing to do the work and submit a pull request for it as well as handling updates as new versions are released if you like. There are a couple of implementation decisions that need to be made first and I might also need you to create a branch or release as I can't do this via pull request. I've added Boards Manager support to 9 different projects as well as contributing to several others and am one of the authors of the Arduino IDE 1.6.x package_index.json format specification.

MCUdude, 2015-12-22:
Awesome! I'd love to have boards manager support! Personally, I'm the guy who builds from source, but using the boards manager makes it a lot easier to get started. I've no experience with Json, nor how it sync with the repository. Does one have to rebuild the Json script every time I commit a change?

per1234, 2015-12-22:
I think it can really increase accessibility to projects like this and hopefully decrease support requests also. The json file will need to be updated with a new entry whenever you want to release a new version for Boards Manager installation. Arduino IDE 1.6.6 and onwards even notifies the user when there are new versions of any installed Board available. The entries for the previous versions will be left in there so that also makes it easy for the user to roll back to a previous version. So ideally you would create a GitHub release whenever you feel a new version is warranted and then I'm willing to handle the update of the JSON file and submit a pull request if you want. I'm watching this repository so I'll get an email whenever you do a release.

MCUdude, 2015-12-22:
Aha! So you'll build a version every time there is a significant bug fix or update? Are these different versions stored on a separate server, or just referring to older repository versions?

per1234, 2015-12-22:
>So you'll build a version every time there is a significant bug fix or update?

The decision of how frequently you want to create a new version is up to you but I'd say that is the general practice. Whenever you want me to add a new version just do a GitHub release or you can give me a mention and I'll go ahead and do it.
>Are these different versions stored on a separate server, or just referring to older repository versions?

Well the tricky thing is that the Boards Manager install archive folder format is slightly different than the folder structure for manual installation(no avr folder). So either you can create a boards manager branch with that folder structure and then merge all commits that you make to the master branch to that branch which will allow the archive files that GitHub automatically generates to be used or I can just manually create the install archive files and they can be uploaded to a branch or as files attached to GitHub releases. These files could also be hosted anywhere else. So the decisions are how to implement the installation archive file and where to store the JSON file.

Here are a couple of examples to demonstrate some options:
[ATTinyCore](https://github.com/SpenceKonde/ATTinyCore) - uses a boardsmanager branch and the releases are based on that branch so the release archive file can be used for the Boards Manager install. The downside to this is that the releases won't work for manual installation but you could create Boards Manager and manual install specific release versions or add other files to the release. The JSON file is hosted on a separate website.

[attiny](https://github.com/damellis/attiny) - uses a boardsmanager branch, the GitHub generated archive file for each commit is used for the Boards Manager install file. There was an issue of the checksum value of one of the archive files changing which breaks the Boards Manager install until the JSON file is updated with the new checksum but that hasn't happened since and it may have been caused by a force push. The JSON file is also located in the boardsmanager branch

Using the automatically generated GitHub archive files has the benefit of the user being able to see exactly what is in the file before installing but adds the burden of maintaining the separate branch.

[AVR_Boot](https://github.com/zevero/avr_boot) - in this case the repository owner wanted the least possible burden of maintaining Boards Manager support so I manually create the Boards Manager archive files and upload them to the gh-pages branch along with the updated JSON file. This way all they need to do is merge a single PR for each release. I like using the gh-pages branch as the JSON file location because it makes a shorter Boards Manager install URL(in this case https://zevero.github.io/avr_boot/package_zevero_avr_boot_index.json).

Other than that the only other decision you would need to make is the name of the JSON file. From the specification:
>The file must be named as follows:
package_YOURNAME_PACKAGENAME_index.json
The prefix package_ and the postfix _index.json are mandatory (otherwise the index file is not recognised by the IDE) while the choice of YOURNAME_PACKAGENAME is left to packager. We suggest to use a domain name owned by the packager for example:
package_arduino.cc_index.json
or
package_example.com_avr_boards_index.json

MCUdude, 2015-12-25:
I've created a v1.0.0 release, and a boards manager branch. In your opinion, what's the best solution? I would like to focus on updating and maintaining the core :)

per1234, 2015-12-25:
It sounds like the system I used for AVR_boot would be best. Is package_MCUdude_MightyCore_index.json for the JSON file name acceptable?

MCUdude, 2015-12-25:
Yes, that JSON file name is fine! The AVR_boot option seems like the one that fits me the most

per1234, 2015-12-25:
I'd prefer if the branch was named `gh-pages` rather than `boardsmanager`. The difference is with the boardsmanager branch the Boards Manager URL will be https://raw.githubusercontent.com/MCUdude/MightyCore/boardsmanager/package_MCUdude_MightyCore_index.json instead of https://MCUdude.github.io/MightyCore/package_MCUdude_MightyCore_index.json. I'm happy to use boardsmanager instead if you want. Since the commit history is irrelevant for this branch you could do:
```
git checkout --orphan gh-pages
git rm -rf .
git commit --allow-empty -m 'Create branch'
```
this will create the branch with only the initial commit. It's not a big deal either way, just a suggestion.

MCUdude, 2015-12-25:
Ok, there's definitely a reason to cal the branch gh-pages, but to be honest, I'm not THAT experienced with github. I created the gh-pages branch, but when I run the $ git rm -rf . command, all my local files gets deleted. How can I clear the content of one branch without deleting all local files?

per1234, 2015-12-25:
When you switch back to master the files should be there it should have only deleted the files in the gh-pages branch if you were in the gh-pages branch when you ran the `git rm -rf .` command which the `git checkout --orphan gh-pages` command should have done.

MCUdude, 2015-12-25:
Of course! Branch created

closed 2015-12-28 by per1234