MCUdude/MightyCore #15: Compiler warning with ATmega324A and ATmega164A

per1234, 2016-03-02:
- Tools > Board > ATmega324 or ATmega164
- Tools > Variant > 324A or 164A
- File > Preferences > Compiler warnings: More or All
- Compile BareMinimum sketch:
```
E:\arduino\hardware\MightyCore\avr\cores\MightyCore_modified\WInterrupts.c: In function '__vector_3':
E:\arduino\hardware\MightyCore\avr\cores\MightyCore_modified\WInterrupts.c:350:12: warning: array subscript is above array bounds [-Warray-bounds]
     intFunc[EXTERNAL_INT_2]();
            ^
```

MCUdude, 2016-03-02:
That's weird. I'm only able to reproduce this error using 324A and 164A. There is a newer version of WInterrupts.c on Arduinos Github repo, but I get the same kind of error. The only difference between the A and the non A versions is the signature IIRC

MCUdude, 2016-03-02:
Looks like the wiring_private.h are missing definitions for 164A/324A. I'm using a [signatures list](http://www.avrfreaks.net/comment/821330#comment-821330) to make sure I don't define the same device twice. Should be OK now

closed 2016-03-02 by MCUdude