MCUdude/MightyCore #16: IRremote library not working

mhdfasilwyd, 2016-03-04:
Pls make Mighty core supported IRremote library

MCUdude, 2016-03-04:
I'll take a look at it, but I'll need to know what kind of microcontroller are you using.

mhdfasilwyd, 2016-03-04:
Iam Using Atmega16 & 32

MCUdude, 2016-03-04:
The library now compiles using ATmega8535, ATmega16 and ATmega32 (They are almost identical except for the memory). Please replace your **IRremoteInt.h** with the one I've attached. I'm going to do a pull request to the IRremote repository so that this fix will be incorporated.

If you're curious about which changes I did, Have a look at line 193/194 and 279/280 :) The only IR pin you can use on these microcontrollers are digital pin 13.

[IRremoteInt.h.zip](https://github.com/MCUdude/MightyCore/files/159069/IRremoteInt.h.zip)

mhdfasilwyd, 2016-03-04:
It is because that this have only one timer

MCUdude, 2016-03-04:
Yes. Timer2 was the default timer, but there are no Timer2A and Timer2B, only Timer2. The ATmega8535/16/32 and the ATmega8 are quite similar when it comes to internal registers, so usually the ATmega8 configuration will work

closed 2016-03-04 by MCUdude

mhdfasilwyd, 2016-03-05:
It worked thanks a lot MCUdude