MCUdude/MightyCore #22: Additional pin out and variant request.

markab, 2016-03-28:
Hello,

I have been looking for a new boot loader and core for my 1284P boards since the old one was broken by 1.6IDE, this looks like it could be a great replacement.

Could you add a pinout variant for octaguino, I have attached the pins_arduino file, and could you make a 16 and 8Mhz variant, I have included my old boards file.

Thanks, looking forward to testing!!

[boards.txt](https://github.com/MCUdude/MightyCore/files/191877/boards.txt)

[pins_arduino.txt](https://github.com/MCUdude/MightyCore/files/191878/pins_arduino.txt)

MCUdude, 2016-03-28:
Hi! It seems like your Octaguino is using the "Standard" pinout. You'll find the pinout option in the tools menu :smile:

markab, 2016-03-28:
[image]

markab, 2016-03-28:
Thanks but the built in led is defined on a different port in my file, is it possible to add?

MCUdude, 2016-03-28:
If you want I can compile a new version of the bootloader where the LED is located at pin 7. The only change you'll need to add to the original pins_arduino.h is located at [line 65 and 66](https://github.com/MCUdude/MightyCore/blob/master/avr/variants/standard/pins_arduino.h#L65). Just replace '0' with '7'.

MCUdude, 2016-03-28:
here's the bootloader files. Replace the old in **avr/bootloaders/optiboot/atmega1284** ones with these

[Optiboot_Octaguino.zip](https://github.com/MCUdude/MightyCore/files/191925/Optiboot_Octaguino.zip)

markab, 2016-03-28:
Thanks thats perfect, all working now :) thanks this saved all my boards!!

MCUdude, 2016-03-28:
Great!

closed 2016-03-28 by MCUdude