MCUdude/MightyCore #23: LiquidCrystal library for ATmega16A

deadbeatfour, 2016-04-28:
Hey, great work with this so far.
I would really love if there was more support for some libraries from the Arduino Playground, specifically LiquidCrystal. Support for NewPing, DS1307RTC, Time and Keypad would be great too.

I really appreciate it!

MCUdude, 2016-04-29:
A lot of libraries doesn't need any modification in order to work. The LiquidCrystal library works with all microcontrollers in this core without any modifications 😄 
[image].

The problems with the "old" AVRs (ATmega8535, ATmega16, ATmega32) is that some of of the register names are slightly different from the new ones (ATmega164, ATmega324, ATmega644, ATmega1284). This means that the NewPing library will compile for the "new" AVRs, while not on the "old" ones. If you want support for a library, you'll have to contact the author in order to get it implemented

deadbeatfour, 2016-04-29:
Well, damn! I feel like an idiot now. Sorry about the bother and thanks for the help :+1: :smile:

closed 2016-04-29 by deadbeatfour