MCUdude/MightyCore #27: Unable to upload files after bootloading 644PA

criticalboot, 2016-06-24:
I used 644 16MHz boot loader on 644PA, worked well last time(3 months back), now I got new 644PA chips now when I tried to boot load from Arduino IDE it show successful burning boot loader message, but "Unable to upload sketches". 

log:
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00

I am trying to understand,
1. How to check boot loader is burned successfully. (Arduino UNO has a led on pin13 which will blink 3 times during boot on power up ).

MCUdude, 2016-06-24:
First turn on verbose compilation and upload in the IDE settings, and post your full error message. It makes it easier for me to help you out. If you want to make sure the bootloader is burnt properly, connect an LED to digital pin 0 (PB0). It should blink twice when you reset the microcontroller

criticalboot, 2016-06-24:
log file attached
[ATMEGA644PA-AU.txt](https://github.com/MCUdude/MightyCore/files/331378/ATMEGA644PA-AU.txt)

criticalboot, 2016-06-24:
I am using TQFP44 not DIP40, does this make any difference? 

LED is connected on PB0, it is not responding, may be bootloading not done.
yes, RX/TX is swapped.

criticalboot, 2016-06-24:
Bootloading log attached.
It shows bootloading down with out errors, no blink on PB0.
[bootloaderlog.txt](https://github.com/MCUdude/MightyCore/files/331571/bootloaderlog.txt)

criticalboot, 2016-06-24:
Through the day today we have been trying various ways to make the 644 bootloader work. Fortunately we were able to boot load it with MicroDuino bootloader. 

I am sure you will able to figure out what has changed between your latest 644 bootloader and the working bootloader which worked for us couple of months ago on 644PA-AU TQFP. We will go ahead and use this for our projects while you may want to figure out and fix the 644 bootloader.

Our observation is that the microduino bootloader is apparently meant for TQFP while we guess that your current bootloader is more applicable on DIP. This is what most likely worked for us.

We would love to use your bootloader once you fix it please let us know.

MCUdude, 2016-06-24:
Please post a picture of your setup. It makes it much easier for me to spot errors. The bootloader should work the same on TQFP and DIP packages. are you using a crystal or an oscillator?

criticalboot, 2016-06-25:
Adding pictures of my setup and pin connections as below.
[picturesof644paausetupwithmightycore.zip](https://github.com/MCUdude/MightyCore/files/332912/picturesof644paausetupwithmightycore.zip)

644PA - AU  --> Arduino UNO
1 --> 11 --> MOSI
2 --> 12 --> MISO
3 --> 13 --> SCK
4 --> 10 --> RESET
5 --> VCC --> VCC
6 --> GND --> GND
7 & 8 pins are connected to crystal (16MHz)
40 connected to led via 330 Ohm resistor.

MCUdude, 2016-06-25:
hmm. It looks correct, except there may be a problem with missing decoupling capacitors. Please try burning the 1MHz bootloader. The clock speed and upload speed is much slower, so it tends to work on some circuits where higher speeds may not work

criticalboot, 2016-06-25:
Thank you for reviewing our work,

Decoupling capacitor : We have actually used as per your suggested minimalistic  circuit (100nf between VCC and GND)  which is probably not visible in the pictures we sent.

Clock Speed : We would love to try with 1MHz boot-loader, but will it work with 16MHz clock speed after boot-loading? The idea is to actually have the final production board to be working at 16MHz. If it works as we want then we are more than willing to try.

However, we are open and willing to try anything that works for us as long as it is practically viable in the production runs.

MCUdude, 2016-06-25:
there's no problem burning the 16MHz bootloader again. Try the 1MHz version and please report back how it goes 😃

MCUdude, 2016-07-11:
Any updates? Did the 1 MHz option work for you?

criticalboot, 2016-08-04:
Sorry for the late reply, 1MHz option also did't work for me.

closed 2016-12-11 by MCUdude