MCUdude/MightyCore #29: Timing with 20Mhz?

mitchsf, 2016-07-18:
Thanks for this core! It's working great, but I'm having a problem with the 20Mhz timing using a 1284P. I switched from Skinny Bob/Bobuino by Christensen, to this, for the 20Mhz option. Testing against a DS3231 that is itself tested against a GPS each minute, the internal clock appears to be off by about one second, each two minutes. The same program works perfectly at 16Mhz using Skinny Bob/Bobuino, with no corrections necessary for the day or two I've kept track.

Any suggestions as to how to narrow this to the root cause, are appreciated.  Thanks again.

MCUdude, 2016-07-18:
I'll have a look at it when I get home. Meanwhile, please try with the 16 MHz option and see if this drifts to. I guess the problem is the millis()  function.  Have you also tried swaping it with another 20 MHz crystal?

mitchsf, 2016-07-18:
Thanks, Hans. Tonight I'll try it with a 16mhz crystal. Several clocks
built with the identical configuration have the same issue, so it's not the
crystal. I agree that it appears to be related to the millis() function.

I'll let you know later after I swap a 16mhz crystal and change the
firmware.

Regards,
Mitch

mitchsf, 2016-07-25:
Still waiting to test. We're moving, so it will be another few days. Thanks.

mitchsf, 2016-07-31:
Sorry. I don't think this is an issue with the core.

closed 2016-07-31 by mitchsf