MCUdude/MightyCore #30: 12 MHz external oscillator with 115200 baud seriel link ?

JensGrabner, 2016-07-20:
Can you in 12 MHz external oscillator  ...   change the seriel link baud to 115200 ?
http://wormfood.net/avrbaudcalc.php/?bitrate=115.2k&clock=12&databits=8
Thanks a lot for do that combination .. i use this for more than 2 years.

MCUdude, 2016-07-21:
Why do you want to use a 12 MHz external oscillator at 115200 baud? The error is 7%!
I will not make this 115200 baud @ 12MHz an option in the boards menu, but you're welcome to compile the bootloader yourself 😉

JensGrabner, 2016-07-21:
Sorry for that mistake. It was my hope that you can set "U2X=1 double speed mode"  --  see the link.

MCUdude, 2016-07-21:
U2X is actually already set, [see bootloader source code](https://github.com/MCUdude/MightyCore/blob/master/avr/bootloaders/optiboot_flash/optiboot_flash.c#L537). The only thing you'll have to do is to edit the boards.txt and compile the bootloader:
`make atmega1284p AVR_FREQ=12000000L BAUD_RATE=115200 UART=0 LED=B0 LED_START_FLASHES=2 -BIGBOOT`

JensGrabner, 2016-07-21:
Thanks a lot. Your "command line" works fine.

closed 2016-07-21 by MCUdude