MCUdude/MightyCore #32: lock/unlock bits fail to verify with Atmel AVRISP mkII and Arduino AVR Boards 1.6.12/avrdude 6.3.0-arduino2

per1234, 2016-08-01:
Tools > Board > ATmega1284P
Tools > Programmer > AVRISP mkII(MightyCore)/AVRISP mkII(they are equivalent for Burn Bootloader)
Tools > Burn Bootloader fails with:
```
C:\Program Files (x86)\arduino-1.6.10\hardware\tools\avr/bin/avrdude -CC:\Users\per\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.6/avrdude.conf -v -patmega1284p -cstk500v2 -Pusb -e -Ulock:w:0x3f:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xf7:m 

avrdude: Version 6.3, compiled on Jun 22 2016 at 16:05:21
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\per\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.6/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : stk500v2
avrdude: usbdev_open(): Found AVRISP mkII, serno: 000200217496
         AVR Part                      : ATmega1284P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
           flash         65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel STK500 Version 2.x firmware
         Programmer Model: AVRISP mkII
         Hardware Version: 1
         Firmware Version Master : 1.23
         Vtarget         : 5.2 V
         SCK period      : 8.00 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e9705 (probably m1284p)
avrdude: erasing chip
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

Error while burning bootloader.
Writing | ################################################## | 100% 0.01s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% 0.00s

avrdude: verifying ...
avrdude: verification error, first mismatch at byte 0x0000
         0xff != 0x3f
avrdude: verification error; content mismatch

avrdude done.  Thank you.
```
The explanation of the cause of this issue is described at https://github.com/arduino/avrdude-build-script/issues/2

This issue will likely also occur for all MightyCore boards, all but ATmega16 have been modified in the patch for this issue, I don't find an entry for ATmega16 in the stock avrdude.conf file. I only have ATmega1284P to test with.

The solution which will still offer backward compatibility with previous Arduino AVR Boards versions requires defining an avrdude tool in the MightyCore Boards Manager JSON file. If you use avrdude 6.0.1-arduino5 then no further changes are required. If you use avrdude 6.3.0-arduino2 then the avrdude.conf patch found at https://savannah.nongnu.org/patch/index.php?8996 will need to be applied and `unlock_bits` and `lock_bits` values in boards.txt must be updated. The avrdude tool will also need to be installed for manual installation.

Another solution would be to just drop support for AVRISP mkII. The Atmel AVRISP mkII is no longer being made but there are still a lot of them around. The Olimex AVRISP mkII is still in production. I have only tested this issue with Atmel AVRISP mkII.

per1234, 2016-08-01:
Same issue with MiniCore(tested with ATmega328P and ATmega168 but all include MCUs are in patch 8996).

Probably same issue with MegaCore(untested because I don't have the microcontrollers but the MCUs are in the patch)

Probably same issue with MicroCore(untested because I don't have the microcontroller but the MCU is in the patch)

MCUdude, 2016-08-01:
Great! Now how do I apply this .patch file to the existing avrdude.conf files? Should there be a simple shell script to do all this, is the file executable from can the file be executed using avrdude or is this a copy/paste job?

MCUdude, 2016-08-01:
I've actually experienced the exact same issue with an STK500 starter kit i got from the university. Uploading worked fine if I used the regular stk500 protocol, bur returned the ´0xff != 0x3f´ error if i used stk500pp.. I [reported a bug on this](https://savannah.nongnu.org/bugs/?46461) last year but there was no response

per1234, 2016-08-01:
I patched Arduino AVR Boards 1.6.12's avrdude.conf with patch 8996 here: https://github.com/arduino/avrdude-build-script/files/392783/avrdude.conf.txt. I had to add the .txt at the end so GitHub would let me attach it to the issue.

This is what I did to apply the patch on Windows, should be similar on other OS but will need a different patch tool. I think git can do it. This was my first time using a patch file so there's probably a better way:

1. Download https://savannah.nongnu.org/patch/download.php?file_id=37105
1. Download and install http://gnuwin32.sourceforge.net/downlinks/patch.php
1. Put the patch file and the avrdude.conf file to be patched in the same folder
1. Rename avrdude.conf avrdude.conf.in
1. Create a file named NEWS
1. Create a file named ChangeLog
1. Run this command: `patch --input=no-lock-byte-read-mask.patch --binary --output=avrdude.conf`

The --binary option was required because of the line endings.

The patches of NEWS and ChangeLog files will fail, but those files needed to be there for the patch to continue.

There were some changelog comments at the top of avrdude.conf that didn't patch correctly, probably because the patch is kind of old. Those should be doable by hand.

MCUdude, 2016-08-01:
I don't think I need to install GNUwin32, since I'm using a mac and got a full blown Unix terminal (with the patch command already there). I'm not home right now, but I'll have a look at this in the evening (should be in the middle of the night where you live).

per1234, 2016-08-01:
I agree, you definitely don't need to install the windows patch tool. I was just listing the process I followed. It's actually the middle of the night right now here(California, USA) but I'm a night owl. Hopefully it will add support for your starter kit also. I think that was the idea behind the change in AVRDUDE's behavior and this patch, to make the handling of the unused bits consistent. It's definitely causing a lot of breakage(see also https://github.com/arduino/Arduino/issues/5175) but hopefully will be an improvement in the long term.

MCUdude, 2016-08-01:
Will this patch break backwards compability with older IDEs that uses avrdude 6.0.1?

per1234, 2016-08-01:
The patched avrdude.conf from Arduino AVR Boards 1.6.12 does break backwards compatibility:
```
avrdude: syntax error at C:\Users\per\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.6/avrdude.conf:1071
```
but that's because line 1071 is:
```
usbpid = 0x2110, 0x2140;
```
Apparently that syntax is not compatible with avrdude 6.0.1-arduino5 but the patch doesn't add any of that, it only changes the lock bit read masks. If I copy the patch ATmega1284P lock bit read mask for ATmega1284P:
```
"x x x x  x x x x   o o o o  o o o o";
```
 to your avrdude.conf line 2430 and change the unlock_bits and lock_bits values for ATmega1284 in your boards.txt to:
```
1284.bootloader.unlock_bits=0xff
1284.bootloader.lock_bits=0xcf
```
which, if I understand correctly, are equivalent to the previous values(0x3f and 0x0f) because only the unused bits are changed.  I can use AVRISP mkII and USBasp to Burn Bootloader with both avrdude 6.0.1-arduino5 and 6.3.0-arduino2. I had previously thought this change was possible only because of a difference in avrdude 6.3 and that you'd have to add an avrdude tool to the MightyCore installation but this indicates that may not be the case and you can stick with the current system of just using whichever avrdude is currently installed for the Arduino AVR Boards version the user has.

Of course that's not a very thorough test, I've only really checked this out with the Arduino AVR Boards 1.6.12/avrdude 6.3.0-arduino2 since I wasn't sure which approach you'd want to take with this.

I tried applying the patch to your avrdude.conf and it doesn't want to work because it's an older version than the one from Arduino AVR Boards I patched.

per1234, 2016-08-01:
It works with Arduino as ISP using avrdude 6.0.1-arduino5 and 6.3.0-arduino2 also. For some reason my USBtinyISP has been failing verification of the bootloader lately so I wasn't able to test beyond that step with that programmer. It works intermittently with 512 byte bootloaders but anything larger almost always fails. I have a couple new ones on their slow way here from China.

MCUdude, 2016-08-01:
Great! I'm about to apply the patch, but does the lock bits in the boards.txt file need to be changed as well?

MCUdude, 2016-08-01:
I'm quite sure I didn't succeed applying the patch. After following your instructions this is the output of the patch command
``` bash
$ patch --input=no-lock-byte-read-mask.patch --binary --output=avrdude.conf
patching file ChangeLog
Hunk #1 FAILED at 1.
1 out of 1 hunk FAILED -- saving rejects to file avrdude.conf.rej
patching file NEWS
Hunk #1 FAILED at 9.
1 out of 1 hunk FAILED -- saving rejects to file avrdude.conf.rej
patching file avrdude.conf.in
Hunk #1 FAILED at 1819.
Hunk #2 succeeded at 2424 with fuzz 1 (offset -902 lines).
Hunk #3 FAILED at 2607.
Hunk #4 FAILED at 2799.
Hunk #5 FAILED at 2991.
Hunk #6 FAILED at 3183.
Hunk #7 FAILED at 3340.
Hunk #8 FAILED at 3542.
Hunk #9 FAILED at 3750.
Hunk #10 FAILED at 3958.
Hunk #11 FAILED at 4152.
Hunk #12 FAILED at 4385.
Hunk #13 FAILED at 4547.
Hunk #14 FAILED at 4729.
Hunk #15 FAILED at 4914.
Hunk #16 FAILED at 5140.
Hunk #17 FAILED at 5331.
Hunk #18 FAILED at 5477.
Hunk #19 FAILED at 5630.
Hunk #20 FAILED at 5787.
Hunk #21 FAILED at 5947.
Hunk #22 FAILED at 6925.
Hunk #23 FAILED at 7137.
Hunk #24 FAILED at 7351.
Hunk #25 FAILED at 7563.
Hunk #26 FAILED at 7753.
Hunk #27 FAILED at 7942.
Hunk #28 FAILED at 8128.
Hunk #29 FAILED at 8310.
Hunk #30 FAILED at 9177.
Hunk #31 FAILED at 9367.
Hunk #32 FAILED at 9575.
Hunk #33 FAILED at 10746.
Hunk #34 FAILED at 10937.
Hunk #35 FAILED at 11141.
Hunk #36 FAILED at 11338.
Hunk #37 FAILED at 11525.
Hunk #38 FAILED at 11714.
Hunk #39 FAILED at 11902.
Hunk #40 FAILED at 12090.
Hunk #41 FAILED at 12244.
Hunk #42 FAILED at 12435.
Hunk #43 FAILED at 13859.
42 out of 43 hunks FAILED -- saving rejects to file avrdude.conf.rej
```

Any ideas? If yours succeeded, maybe you could upload it here so I can add that one instead?

per1234, 2016-08-01:
I had the same problem. It's because your avrdude.conf is too old. The patch was successful on the Arduino AVR Boards 1.6.12 ardude.conf because it's more up to date. Here's the options I see:

1. You could use the patched file Arduino AVR Boards 1.6.12 ardude.conf and then merge back in your modifications. That file has the syntax incompatible with avrdude 6.0.1-arduino5 which will cause incompatibility with previous Arduino AVR Boards versions so you will also need to add an avrdude 6.3.0 tool to MightyCore. From cmaglie's recent statements it sounds like the current system of using Arduino AVR Boards tools is not recommended so that may be required anyway.

1. Manually copy the modified lock read masks from the patch to your current avrdude.conf which, from the test I described above, appears to retain compatibility with avrdude 6.0.1-arduino5.

I'm happy to help out with any of this but my ability to test is limited to the MCUs I have on hand(ATmega1284P, ATmega328P and ATmega168 are the ones relevant to your cores).

MCUdude, 2016-08-01:
I think option 2 is the best choice. A lot of users are still using older versions of the IDE. I've very little (ok, no experience) with the patch command, how it works (identifying what to change and what to leave) and how it can be done manually. I'd really appreciate some help with this one 😃 After the patched file is added I should be able to do this on the other cores as well.

If I'm a bit late to reply is it because I'm on vacation, and doesn't got access to a computer all the time. I get all the notifications on my phone though 😉 
Thanks for being such a resource for me and the rest of the community. It's people like you that make sure code and cores like this don't get obsolete because the Arduino team always seems to break backwards compatibility with every update.

per1234, 2016-08-01:
Option 1 won't break backwards compatibility because avrdude 6.3.0-arduino2 will be installed with MightyCore regardless of the IDE version. The downside is that this means a larger install and duplicate tools. I'm hoping the Arduino developers will come up with a way that tools can be shared between cores without causing the sort of problems we're experiencing now but this will be tricky and any fix will only apply to future IDE versions. The MightyCore specific tools approach will be more of a hassle for manual installation because you will have to either include the tools in this repository to allow a installation via a single download or instruct the users to install the tools separately, adding to the complexity of the installation process.

This situation has definitely been a project since I have quite a few different 3rd party boards projects I'm involved with. I haven't done any work on some of them yet because I'm still trying to figure out the best approach to fixing all this breakage while maintaining backwards compatibility. Throw in Eclipse Arduino plugin compatibility, etc. and it gets even more complex.

MCUdude, 2016-08-01:
The absolute best option would be to make sure all users runs the same version of avrdude. If the first option won't break backwards compatibility and force users to use "my" version of avrdude (in this case avrdude 6.3) then I think we should stick with this.  I'm not a conservative person myself, and I think that new and better software is great! 👍 IIRC the avrdude executable
 is less than a megabyte, so duplicated 'dudes is what I'd call "splitting hairs" when it comes to space. I'll agree it's not that elegant though..

When it comes to the tool dependencies I've decided to wait until the dust settles. There's a long going discussion, and dealing with the Eclipse add-on makes it even harder. Hopefully there will soon be a fix we all can agree on.

per1234, 2016-08-02:
I agree that is the best option. Maintaining backwards compatibility is very important in my opinion and I try to consider it with anything Arduino related I do. avrdude 6.3.0-arduino2 has already fixed one bug(https://github.com/arduino/Arduino/issues/2986) I know of and I suspect another one as well(https://github.com/arduino/Arduino/issues/388). It seems to be incompatible with USBasp using libusb-win32 drivers(https://github.com/arduino/avrdude-build-script/issues/1) but that can be fixed by changing to a different driver and I'm hoping they will be able to fix it since the stock AVRDUDE 6.3 doesn't have that issue..

I'll work on this today and let you know when I have something ready for testing.

per1234, 2016-08-04:
Manual installation file for testing: https://github.com/per1234/MightyCore/archive/no-lock-byte-read-mask.zip

Boards Manager URL for testing: https://raw.githubusercontent.com/per1234/MightyCore/add-avrdude-630-tool/package_MCUdude_MightyCore_index.json

Note that similar changes to Arduino AVR Boards are being considered in https://github.com/arduino/Arduino/pull/5202 but the developer has put it on hold while other options are considered due to the breakage it will cause for 3rd party boards. That's not an issue for MightyCore but it could be useful to see what solution Arduino comes up with.

### Changes
- Added the avrdude 6.3.0-arduino2 tool to MightyCore 
- Updated MightyCore's avrdude.conf to avrdude 6.3.0-arduino2
- Applied changes from patch 8992 to avrdude.conf
- Update boards.txt to reflect lock bit read changes caused by patch 8992
- Update platform.txt to use MightyCore's avrdude tool

### I tested:
- Manual and Boards Manager installation
- ATmega1284P
- USBasp, Arduino as ISP, Atmel AVRISP mkII and USBtinyISP(won't verify the hex files so it never gets to the `lock_bits` verification but it did verify `unlock_bits` value but this issue is not isolated to this fix, any core, or any version of avrdude).
- Manual installation back to Arduino IDE 1.6.2, the earliest previously compatible version.
- Boards Manager installation back to Arduino IDE 1.6.4, the earliest version that supports 3rd party Boards Manager installations.

### Not tested:
- Any MCU other than ATmega1284P
- `lock_bits` verification with USBtinyISP
- Linux, Linux 64 bit, Mac OSX, ARM Linux

### Issues:
1. There are 5 different builds of avrdude for each operating system. I added all of these to the repository adding >9 MB to the manual installation size. Boards Manager handles all this automatically so this issue only relates to manual installation and the master branch of the MightyCore repository.
3. The [3rd party hardware specification](https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5-3rd-party-Hardware-specification#global-predefined-properties) only specifies `{runtime.os}` values `linux`, `windows`, and `macosx` but there are 5 different avrdude builds. I found `linux64` used in the [Arduino-STM32 core](https://github.com/rogerclarkmelbourne/Arduino_STM32/tree/master/tools). So that leaves the "arm-linux-gnueabihf" build. I took a wild guess that this value is `linuxarm` because that's what Arduino uses for their test build names(e.g. https://github.com/arduino/Arduino/pull/5202#issuecomment-237322962). I posted a [question ](http://forum.arduino.cc/index.php?topic=416545)about this to the forum but no response yet. If anyone reading this has a Raspberry Pi, etc. running Arduino IDE and is willing to check it out let me know, I can provide a test boards package and I'll add the information to the specification.
7. git didn't like the symlinks in the linux, linux64, linuxarm and macosx avrdude folders. I was able to fix this with `git config --bool core.symlinks true`.
5. With the Boards Manager installation I can use for `{runtime.tools.avrdude-MightyCore.path}` for `tools.avrdude.path` and `tools.avrdudeCustom.path` but that doesn't work for the manual installation, I have to use `{runtime.platform.path}/tools/{runtime.os}/avrdude`. This means there needs to be a separate version of platform.txt for each installation type.

### Potential changes
A possible solution to issues 1-4 is to create a release script that downloads MightyCore and each build of avrdude and assembles an installation file for each OS which would be attached to the GitHub release. This would eliminate the need to have the avrdude tool in an OS specific folder so the use of `{runtime.os}` in platform.txt would no longer be necessary. The avrdude files would no longer need to be added to the MightyCore repository which fixes the symlinks git issue. This could also handle the modification of platform.txt to work with Boards Manager, create the Boards Manager installation file, and update the JSON file. Some of this is demonstrated in: https://github.com/Optiboot/optiboot/blob/master/optiboot/release.sh. I have no experience with this sort of thing and am not really motivated to learn. Anybody out there have experience in this area that wants to help out?

If you want the tools folder organized differently that's doable, it just needs to be inside the avr folder, other than that there's nothing fixed, it doesn't even need to be in a tools folder, though that makes sense for consistency.

I needed to give the Boards Manager installation's avrdude a unique name to avoid https://github.com/arduino/Arduino/issues/5042. This issue has been fixed in the hourly build but that doesn't help for backwards compatibility. The current tool name is `avrdude-MightyCore` but it can be changed to anything you like as long as it's a unique name that won't ever be used by any other core.

MCUdude, 2016-08-04:
Thanks! I'll be home on sunday, so I will test all this when I get home. If everything works fine I'll add it to the repo 😃

per1234, 2016-08-04:
An update on Arduino's approach to this issue(https://github.com/arduino/Arduino/pull/5202): The developer was concerned about the breakage of 3rd party boards that it would cause so has completely reworked the PR with a patched avrdude what allows the stock fuse settings to verify. I haven't tested it yet. If that change is merged it would mean that the issue described here will only affect users with Arduino AVR Boards 1.6.12 installed, any previous or newer versions will work with MightyCore as it is now.

I'm no 100% clear on why the behavior of avrdude was changed between 6.0.1 and 6.3 but there must have been good reason for making such a large change. I believe the motivation was to make the handling of unused bits consistent. Arduino's approach seems like a step backwards to me.

The problem this will cause for MightyCore if my proposed changes are accepted is that we will be unable to use new avrdude versions from Arduino so will be either forced to always use avrdude 6.3.0-arduino2, build avrdude from source, or use the prebuilt stock avrdude images which don't have Arduino's other patches.

MCUdude, 2016-08-07:
Thanks! I'll tested your "patched" core (manual installation), and it works fine with IDE 1.6.10. I burned the bootloader and uploaded a simple sketch to an ATmega32. I noticed though that the IDE chose gcc 4.8.1 instead of the latest one I hope the Arduino devs are working on a solution that doesn't lock the user into having to use the patched avrdude that comes with the IDE. It's a fix, but a rather dirty one. Maybe the best solution would be to include the tools folder, like you have done?

MCUdude, 2016-08-07:
what if we include the stock/vanilla version of avrdude in this and the other cores, and modify the lock bits in the boards.txt file to work with the standard version of avrdude? Then we don't need to care about what the Arduino devs are doing!

per1234, 2016-08-07:
>I noticed though that the IDE chose gcc 4.8.1 instead of the latest one.

That's not caused by this version of MightyCore. Now that the toolsDependencies entries for avr-gcc 4.8.1 have been removed, installing MightyCore will not cause it to be installed. However, if that version has been installed by a different core then it will always be chosen by the IDE, for MightyCore, Arduino AVR Boards, etc.

> Different versions for manual and boards manager seems a bit unfortunate and not very elegant.

It's a minor difference in two lines of platform.txt but I agree. There is no documentation of how to add tools to a manual installation. The only core I found that used tools was Arduino-STM32 so I copied how they did it but they don't have a Boards Manager installation so it isn't an issue for them. This thread: https://groups.google.com/a/arduino.cc/forum/m/#!msg/developers/TDYnunLEqXU/w6jRV_FcBAAJ indicates there is a way to make it so that a single platform.txt can work for Boards Manager and manual installation:
> >Paul Stoffregen ha scritto il 18/09/2015 alle 15:25:
> > I can't seem to get {runtime.ide.path} to work at all in the new
> > builder.  I do need some way to construct a {compiler.path} that is the
> > location where I install my customized toolchain.....
> >
> 
> There are two ways
> 
> First one (preferred) is to create a folder structure such as
> "tools > name of tool > version of tool > every other file"
> 
> This is what boards manager does. For example
> 
> tools/bossac/1.6-arduino
> 
> This will create runtime.tools.bossac.path and runtime.tools.bossac-1.6.path
> 
> Second one is to put a builtin_tools_versions.txt file where your tools
> are installed. That's what the IDE you download does, because it bundles
> the AVR toolchain
> You can find that file in IDE/hardware/tools/avr/ 

But I haven't managed to get that to work. No matter where I put the tools folder and whether I add the builtin_tools_versions.txt or not {runtime.tools.avrdude-MightyCore} never gets defined. I think I just need to keep at it. It's very frustrating how poorly documented the Arduino IDE is. I shouldn't have to dig some comment up on a mailing list and do a zillion trial and errors to figure out something that the developer, who knows exactly how it works because he wrote the code, could have spent 2 minutes adding to the documentation. There are so many hidden features like this. The guy who wrote half this stuff doesn't even work for Arduino any more so there are probably features nobody knows about.

> All I want is the core to work with the latest version of the IDE without having to rely on some magical patch to make this working.

Actually that's what the Arduino developers are planning. They are going to revert to avrdude 6.0.1-arduino5 for the soon to be released [IDE 1.6.11/Arduino AVR Boards 1.6.13](https://github.com/arduino/Arduino/pull/5208) and then patch the avrdude 6.3.0 source so that it will work with the standard Arduino fuse settings and release that in Arduino AVR Boards 1.6.14. They're going to try to get the changes merged upstream in avrdude. I think the first attempt is done here: https://github.com/arduino/Arduino/pull/5202#issuecomment-237621343 but it sounds like there are going to be some significant changes from that implementation. The discussion of the decision to revert the avrdude version here: https://github.com/arduino/Arduino/pull/5021#issuecomment-237844573.

I'm not sure what the Arduino developers plan to do to avrdude to make this work. I have to assume that the changes that the avrdude developers made between 6.0.1 6.3.0 were thoroughly considered since they represent such a large change. The purpose change the avrdude developers made was to add consistency with Atmel's specifications, I just get the feeling that any workaround the Arduino developers make will basically be working backwards away from that, essentially a hack, and won't have a chance of being merged upstream. I hope I'm wrong.

>Ideally this core should work with stock avrdude 6.0.1 and 6.3

Well it depends on your definition of "work", right now the only thing broken is AVRISP mkII(and probably other stk500v2 programmers) support when used with one Arduino AVR Boards 1.6.12. This issue will probably not occur with future Arduino AVR Boards versions. So this issue is fairly limited in scope and the problems associated with fixing it may outweigh the benefits. On the other hand, the changes described here offer the added benefit of allowing you complete control of the tools your core is used with, instead of being at the mercy of whatever Arduino decides to unleash. This is the first step towards getting Eclipse support back, an avr-gcc tool would be the next step. The Eclipse plugin developer got a patreon sponsor to pay him to investigate the issue and took the time to actually understand the problem. He may decide he has no choice but to adapt to the solution that the majority of 3rd party boards have settled on(removing the toolsDependencies items).

MCUdude, 2016-08-07:
> On the other hand, the changes described here offer the added benefit of allowing you complete control of the tools your core is used with, instead of being at the mercy of whatever Arduino decides [...]

You have a point! I'd rather include my own version of avrdude and avr-gcc and not having to rely on what the Arduino team choose to do. I'd always struggle to make sure the core and the core tools (avrdude, avr-gcc) is up to date, but I can add these in my own tempo.

MCUdude, 2016-08-07:
So how do you think we should solve the different platform.txt issue for manual/boards manager install? Is it best to use the gh-pages to store all the boards manager related stuff, or should I create a new branch? It's easy to forget updating both branches when their content is almost identical. Is it possible to "link" two branches, e.g I update a file in one branch, and the file gets automatically updated in the other fork?

per1234, 2016-08-08:
The Windows version of avr-gcc 4.9.2-atmel3.5.3-arduino2 is 42MB compressed. I'm guessing the other versions are in the same ballpark. With the current manual installation system I'm using that would make for a much larger download. This would be improved by the release script I mentioned previously. Boards Manager will only download the version for the user's operating system and if the same version of avr-gcc has been previously downloaded then it will use the copy stored in Arduino15/staging/packages instead of downloading it again.

While the release script would be the best system, another option is to manually assemble the installation files for each operating system to attach to the release. Unfortunately adding files to a release can't be done via pull request so that responsibility would fall on your shoulders. If you're not up to writing the script you could make an issue for it with a "help wanted" tag and maybe put the word out on the forum threads for each core also and then use the "wetware" script until some kind soul comes along to help out.

>I'd always struggle to make sure the core and the core tools (avrdude, avr-gcc) is up to date, but I can add these in my own tempo.

That's actually a big advantage of MightyCore having its own tools. With the current system of using whatever tool versions Arduino AVR Boards is using you have to do a bunch of testing as soon as Arduino updates to new versions. I think it was less than a day between the tools being added and the release of Arduino AVR Boards 1.6.12. The first you know of an incompatibility might be one of the users reporting the issue. With MightyCore specific tool versions you can update the tools when you are ready, test them at your own pace, and only release the new version when everything is in a fully working state.

avr-gcc 4.9.2-atmel3.5.3-arduino2 has LTO but with the current system of using whatever version of Arduino AVR Boards tools the user has installed adding the flags to platform.txt to enable LTO will break backwards compatibility. If MightyCore had its own avr-gcc 4.9.2-atmel3.5.3-arduino2 tool then you would know that enabling LTO will work regardless of what Arduino AVR Boards or IDE version the user has.

>So how do you think we should solve the different platform.txt issue for manual/boards manager install?

I'm going to systematically work through every possible combination of tool folder location and structure today until I can get the IDE to define `{runtime.tools.avrdude-MightyCore}` for the manual installation. If I can figure that out the identical platform.txt will work for both.

per1234, 2016-08-08:
I've come to the conclusion that there's no way to get {runtime.tools...} values set for manually installed boards. It relies on the tool entries in the Boards Manager JSON files. Along the way I figured out a different way to use the same path value for Boards Manager and manual installations and thus not need to have different platform.txt files:
`{runtime.hardware.path}/../../tools/avrdude-MightyCore/6.3.0-arduino2`
It's not pretty but it does work.

The Boards Manager avrdude-MightyCore tool is located in the same place but manual install avrdude-MightyCore was moved to `{sketchbook folder}/tools/avrdude-MightyCore/6.3.0-arduino2`. I think this might be a bit of an abuse of that folder because I've previously only seen it used for IDE tools(like https://github.com/me-no-dev/EspExceptionDecoder), rather than hardware tools, but it does work. 

Since I can't use the `{runtime.os}` value anymore in the manual installation path separate installation files will need to be created for each OS(as I've done for the test installation files). I think that would have ended up being the best system anyway. The disadvantage of not having the tools included in the repository is people can no longer just download https://github.com/MCUdude/MightyCore/archive/master.zip to do beta testing. They will need to manually install the tool(s) also(unless they already have the right version installed from a previous MightyCore installation) but I can write some instructions for how to do that.

The manual installation instructions have changed a bit since there are now two separate folders under the sketchbook affected.

Since the path value in platform.txt contains the tool version it will need to be updated whenever you update to a new tool version. The tool download URLs in the beta testing instructions would also need to be updated.

### Boards Manager URL(v1.0.7-test2 only):
https://raw.githubusercontent.com/per1234/MightyCore/add-avrdude-630-tool2/package_MCUdude_MightyCore_index.json
### Manual installation files:
- [MightyCore-1.0.7-test2-linux32.zip](https://github.com/MCUdude/MightyCore/files/406456/MightyCore-1.0.7-test2-linux32.zip)
- [MightyCore-1.0.7-test2-linux64.zip](https://github.com/MCUdude/MightyCore/files/406455/MightyCore-1.0.7-test2-linux64.zip)
- [MightyCore-1.0.7-test2-linuxarm.zip](https://github.com/MCUdude/MightyCore/files/406457/MightyCore-1.0.7-test2-linuxarm.zip)
- [MightyCore-1.0.7-test2-macosx.zip](https://github.com/MCUdude/MightyCore/files/406458/MightyCore-1.0.7-test2-macosx.zip)
- [MightyCore-1.0.7-test2-windows.zip](https://github.com/MCUdude/MightyCore/files/406459/MightyCore-1.0.7-test2-windows.zip)

### Manual installation instructions:
1. Download the installation file for you operating system.
2. Move the downloaded file to your sketchbook folder.
3. Unzip the file.

per1234, 2016-08-11:
It turns out there is an issue(https://github.com/arduino/Arduino/issues/5173) with Arduino's build of avrdude 6.3.0-arduino2 that causes upload to fail when the IDE is started by opening a sketch(I think Windows only). I've been unable to reproduce this issue(may be Windows 10 only and I'm on Windows 7) and am not sure if it would occur with a tool installed via Boards Manager rather than the version included with the Arduino IDE. I saw a report on the forum today of someone experiencing the issue with Arduino IDE 1.6.9/Arduino AVR Boards 1.6.12 which makes me think there is a possibility the issue would affect a MightyCore avrdude 6.3.0-arduino2 tool.

The flaw was caused by Arduino's build accidentally having an external dll dependency which is not found because starting the IDE by opening an ino file doesn't add the correct IDE folder to the path. I suspect the problem could be fixed by just adding that file(libusb-0.dll) to the tool folder but I can't confirm this since I can't reproduce the issue.

If anyone reading this can reproduce https://github.com/arduino/Arduino/issues/5173 and is willing to help out please comment.

If the dependency issue can't be solved it doesn't necessarily rule out the idea of adding a MightyCore avrdude tool, I see a few alternative tool options:
- Stock avrdude 6.3, available prebuilt from http://download.savannah.gnu.org/releases/avrdude.
  - Need to determine whether the stock avrdude will cause https://github.com/arduino/Arduino/issues/5173.
  - Doesn't have Arduino's custom patches(https://github.com/arduino/avrdude-build-script). I have no clue what most of that does. The only two patches I understand affect avrdude.conf and thus are irrelevant since MightyCore has it's own avrdude.conf file already based on Arduino's version.
- MightyCore's own build of avrdude 6.3.0 using the Arduino patches but with the dll dependency issue fixed. The Arduino developers should add a commit for that before releasing the next avrdude version.
- Wait for the new Arduino avrdude version to be released(avrdude 6.3.0-arduino3 or whatever).
- avrdude 1.0.6-arduino5: This would fix the AVRISP mkII lock bits issue when MightyCore is used with Arduino AVR Boards 1.6.12 and it is a proven version but we lose the advantage of any improvements that have been made since that version.

MCUdude, 2016-08-11:
I'm sorry I've been so late to reply, I'm still on vacation 👍 I'll have a look at it when I get home on Monday

per1234, 2016-08-11:
No worries, just enjoy your vacation! I don't think it's that critical anymore if the Arduino developers will just release Arduino IDE 1.6.11/Arduino AVR Boards 1.6.13, more of a long term possibility to consider. Even if only a learning exercise it's taught me what will be required to add LTO support to cores without breaking backwards compatibility, which offers more potential advantages than the avrdude update.

MCUdude, 2016-08-18:
BTW Arduino IDE 1.6.11 is out! It seems like they rolled back to Avrdude 6.0.1

per1234, 2016-09-23:
The release of Arduino AVR Boards 1.6.14/Arduino IDE 1.6.12 has brought the return of AVRDUDE 6.3. The new avrdude 6.3.0-arduino6 is patched to deal with the issue of changed unused fuse bits handling causing incompatibility with previous fuse settings. Now when you burn bootloader to ATmega1284P with AVRISP mkII programer it verifies the lock bits successfully but with the following warning:
```
avrdude: WARNING: invalid value for unused bits in fuse "lock", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0xcf instead of 0x0f (double check with your datasheet first).
```
So this is a great improvement over avrdude 6.3.0-arduino2 but I have already seen multiple reports from users scared by this warning(people frequently don't understand the difference between a warning and an error) in the 2 days since the release so I'm guessing you will be hearing about this from time to time.

I'm not sure how earnest the Arduino developers are about eventually removing backwards compatibility for old fuse values or how soon that might happen. It may depend on whether something like their patch is merged upstream by AVRDUDE.

MCUdude, 2018-02-20:
Just going through some old issues here. Is there anything that needs to be done in order to close this?