MCUdude/MightyCore #33: Add additional Wiring functionality

MCUdude, 2016-08-13:
The Wiring core (which the Arduino core is based on) includes a lot of additional functions and methods that the Arduino team never bothered to include to the Arduino core. Functions like:
* enablePower()
* disablePower()
* noAnalogWrite()
* setPWMPrescale()
* setPWMResolution()
* pullup()
* noPullup()
* portMode()
* portWrite()
* portRead()
* sleep()
* noSleep()
* sleepMode()

Are really useful for the more experienced Arduino user, and shouldn't be hard to implement. What do you guys think? 😃 I think I'll have to add some additional reference to these extra functions.

Oh, if you're interested in some Arduino vs Wiring history, I recommend you to read this "article" about this turbulent relationship: [The untold History of Arduino](https://arduinohistory.github.io).

MCUdude, 2016-08-17:
It didn't turn out to be hard at all! Issue closed. e5c50ea

closed 2016-08-17 by MCUdude

per1234, 2016-08-17:
I was thinking just last night about adding support for the Wiring boards to a boards package I'm in charge of. The Wiring S is an ATmega644P and the Wiring 1.0 is a ATmega128, they made ATmega1281 and ATmega2561 boards also. I haven't written one before but it shouldn't be too hard to make a pins_arduino.h file for them. I was thinking of doing it mostly as a nod to the history behind Arduino but there must be quite a few of those boards out there(you can still buy the Wiring S) and the Wiring IDE seems completely dead.

Any interest in adding those variants to MightyCore and MegaCore? I'm not sure if it would be a good idea or not, there's a lot to be said for keeping things simple. It's really unfortunate that there are so many different pin mappings. I have a list of 12 unique variants for ATmega1284P.

MCUdude, 2016-08-18:
I don't think it's a good idea to add these boards to MightyCore and MegaCore, and here's why: First, I don't believe the Wiring boards are very popular compared to Arduino. No one would probably use this extra functionality anyway. This also means that there would be even more pinouts. I like to keep things simple, so I think I'll stick with the most used pinouts (which is the standard and the Bobuino pinout i believe).

However, I've planned to add support for the ATmega1281 and ATmega2561 (MegaCore), since the pinout is practically identical to the ATmega64/128 pinout. The only thing that stops me is some actual chips. I have no idea why, but Atmel won't send me free samples anymore.. 😞