MCUdude/MightyCore #35: Can upload program code only one time, then I have to burn bootloader again

Quadbow, 2016-08-31:
Hi Hans (MCUdude),
I have some ATMega32 I'd like to use with Arduino. So, I tried out MightyCore with an own board, connecting it to USB via an Arduino Promini FTDI adapter and to the hardware USART pins.
There is no problem to install MightyCode under Windows7.
There is no problem to burn the relevant Optiboot bootloader from the local MightyCore path.
There is no problem to select the ATMega32 MCU, to compile it and to run it for the first time.
I tried out the blink example programm - everything is fine - at least at the first glance.
But, after uploading the sketch, the LEDs of the FTDI adapter (which are usually dark when not uploading) are now slightly shimmering. And - even worse - the system refuses to upload any sketch afterwards. 
In short: I only can upload any sketch, when the bootloader was burnt freshly. This is valid for all versions of MightyCore, since I tested all from 1.0.0 to 1.0.6.
The setting of the fuses are: SPIEN, BOOTSZ=256W_3F00, BOOTRST, external xtal.
I would be glad, if you could give me an advice how to fix that issue.
Thanks in advance - Quadbow.

MCUdude, 2016-08-31:
The reason why you can upload the sketch just once is because the your auto reset isn't working. After you burnt the bootloader, the microcontroller is resetting itself every second or so, so uploading a sketch will always work the first time. You should probably check your reset circuity

Quadbow, 2016-09-01:
Hi and thanks for the fast response. I was not aware about the bootmanager's self resetting. So, I suspected a software issue. With your explanation it seems now to me, that there is a hardware issue. I will check and find out and report back.

MCUdude, 2016-09-03:
Did you figure this out?

Quadbow, 2016-09-07:
I found two faults. The first was a a bad soldering and the second wrong parameters for the fuses.
So, the issue can be closed. It had really nothing to do with your code.

closed 2016-09-07 by Quadbow