MCUdude/MightyCore #36: 18.4320MHz clock

Jimmy-Jeong, 2016-09-07:
Hi. I have tested with Ethernet shield and it's working very well.
Thank you for supply this boot-loader
And, I have a question.
Do you have plan to update 18.4320MHz clock option??

MCUdude, 2016-09-07:
Hi!
IIRC the benefit of using such a clock frequency is "error free" serial -> [Wormfood's baud rate calculator](http://wormfood.net/avrbaudcalc.php?bitrate=300%2C600%2C1200%2C2400%2C4800%2C9600%2C14.4k%2C19.2k%2C28.8k%2C38.4k%2C57.6k%2C76.8k%2C115.2k%2C230.4k%2C250k%2C.5m%2C1m&clock=18.432&databits=8)

I'm not planning to add 18.4320 MHz to this core, because timing (millis, micros and delay) will be completely off. You can always fork this repo and add the 18.4320 MHz bootloader yourself if having this frequency is important 👍

MCUdude, 2016-09-07:
Oh, don't hesitate to ask if you need help to compile the bootloader or implement it to the core! 😉

Jimmy-Jeong, 2016-09-08:
Thank you for your reply and good information.
I will ask if I have a queasion. :)

closed 2016-09-08 by Jimmy-Jeong