MCUdude/MightyCore #41: get rid of warnings

zevero, 2016-11-08:
just remove the const on returning by value :)
```
/home/severin/.arduino15/packages/MightyCore/hardware/avr/1.0.7/libraries/EEPROM/src/EEPROM.h:43:30: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
     operator const uint8_t() const       { return **this; }
                              ^
/home/severin/.arduino15/packages/MightyCore/hardware/avr/1.0.7/libraries/EEPROM/src/EEPROM.h:92:26: warning: type qualifiers ignored on function return type [-Wignored-qualifiers]
     operator const int() const          { return index; }
                          ^
```

MCUdude, 2016-11-08:
@Chris--A will this affect the code in any way?

MCUdude, 2016-11-08:
Hmm. I'm one getting an error when doing that. Is is "safe" @Chris--A ?

zevero, 2016-11-08:
I am sorry ... its an issue with the EEPROM library ... not with MightyCore

closed 2016-11-08 by zevero

MCUdude, 2016-11-08:
No problem! I include the library in the core, so I'll update it if @Chris--A have a proper fix for it 😉

Chris--A, 2016-11-09:
Yeah I have an update in the mix. This and other warnings are fixed.
Also have a few more additions.
Been flat out the last few weeks, will do a PR on Friday/Saturday
Cheers