MCUdude/MightyCore #42: problem with bulid in sloeber for all MightyCore

yoavshtainer, 2016-11-22:
first of all, thank you for your great work!

i'm using atmega644 and atmega1284 with arduino ide with no problems.

but in the last week i  decided to work with the sloeber and i cant build any project for the MightyCore

i'm a rookie in eclipse so i don't understand the hole thing.

my error is:

`13:58:54 **** Clean-only build of configuration Release for project dsfsdf ****
"C:\\sloeber8\\sloeber\\arduinoPlugin\\tools\\make\\make" clean 
make: *** No rule to make target 'clean'.  Stop.

13:58:55 Build Finished (took 758ms)

13:58:56 **** Build of configuration Release for project dsfsdf ****
"C:\\sloeber8\\sloeber\\arduinoPlugin\\tools\\make\\make" all 
'Building file: ../.ino.cpp'
'Starting C++ compile'
"/bin/avr-g++" -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega1284p -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR   -I"C:\sloeber8\sloeber\arduinoPlugin\packages\MightyCore\hardware\avr\1.0.7\cores\MightyCore" -MMD -MP -MF".ino.cpp.d" -MT".ino.cpp.o" -D__IN_ECLIPSE__=1 -x c++ "../.ino.cpp" -o ".ino.cpp.o"  -Wall
subdir.mk:24: recipe for target '.ino.cpp.o' failed
process_begin: CreateProcess(NULL, sh.exe -c "/bin/avr-g++ -c -g -Os -std=gnu++11 -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -MMD -mmcu=atmega1284p -DF_CPU=8000000L -DARDUINO=10609 -DARDUINO_AVR_ATmega1284 -DARDUINO_ARCH_AVR -IC:\sloeber8\sloeber\arduinoPlugin\packages\MightyCore\hardware\avr\1.0.7\cores\MightyCore -MMD -MP -MF.ino.cpp.d -MT.ino.cpp.o -D__IN_ECLIPSE__=1 -x c++ ../.ino.cpp -o .ino.cpp.o -Wall", ...) failed.
make (e=2): The system cannot find the file specified.

make: *** [.ino.cpp.o] Error 2

13:58:57 Build Finished (took 1s.110ms)`

i don't know how to look about the board.txt file or understand what going on in there, so if you can help me i will be very happy.

thank a lot!

MCUdude, 2016-11-22:
I suggest asking this question at the [Sloeber Github page](https://github.com/jantje/arduino-eclipse-plugin) as this probably is an Eclipse related problem.

yoavshtainer, 2016-11-22:
i will ask there, but i dont have problems with the uno and esp8266. so i think it's related to here.
thank you!

closed 2016-12-11 by MCUdude