MCUdude/MightyCore #43: New Board!

peekpt, 2016-12-05:
Just to let you know that I'm using your Mightycore pinouts standard  to make a board like arduino pro mini 
https://github.com/peekpt/mightyduino 
Please take a look. (The board is ready, I'm waiting for it)
Nice Regards,
Paulo

MCUdude, 2016-12-07:
Looks great! Did you use Kicad for the project?
One tiny little error though; the built-in LED (the one that flashes when you hit the reset button) is connected to D0 and not D13 as your board indicates

peekpt, 2016-12-07:
I wasn't aware of that behavior and use the pin 13 like arduino pro mini.
Why it blinks? Is it on bootloader, or it's an hardware thing? I'm still waiting for the boards, I will correct this on v1.1.

Yes I use kicad and I used to leave on Eagle but when you get into it it's very easy to operate. I can't leave without realtime DRC routing and tracks Shove feature. I was told that  Eagle is more user friendly to send to the maker exporting like smd rotations, glue points, special parts... but kicad for Protoboards it's more than ok.

I hope you don't mind that I'm using your "Mighty" signature ( is like a tribute for your work). 
I did this board because I work with 644p on my personal projects and I needed something small and quick to mount, with minimum amount of components and without ttl-serial. I am using breakouts boards for prototyping 644p.

MCUdude, 2016-12-07:
It's the bootloader that's causing the LED to blink. I'm a Eagle user myself, but is currently learning Altium Designer at work. I'm amazed how non-intuitive Eagle is compared to Altium.

The "Mighty" name have been around singe Maniacbug created the first [Mighty-1284p core](https://github.com/maniacbug/mighty-1284p) about 5 years ago. I didn't understand why no none had created a core that included the allmost-identical 644p, 324p and 164p, so I did it myself 😉

peekpt, 2016-12-07:
Althium is the mainstream but not for all wallets, there's a less featured
expensive version and a free cloud based version though.

I agree, eagle is non intuitive sometimes frustrating.

peekpt, 2016-12-07:
I just update it to 1.1 and readme. It looks good. Thank you for your work

closed 2016-12-11 by MCUdude