MCUdude/MightyCore #46: why nothing happened after uploading a code

AndroidDesigner, 2017-01-20:
hi.
I followed your directions and could burn boot-loader for my ATMega32A and led in pin 1 had blink.
now want to upload a sketch in my ATMega32A.
I wrote a simple blink sketch and set led_pin = 22 and connect a led to pin22 in ATMega32A in breadboard and upload sketch by holding shift key (upload by Programmer).
But nothing happened and led didn't have blink. why?

MCUdude, 2017-01-20:
Most likely something wring with your code. Try uploading the standard blink.ino sketch. If the LED connected to pin 0 (PB0) togges every second, there's nothing wrong with the MightyCore installation

AndroidDesigner, 2017-01-20:
thank you very much for rapid reply.
yes I tested standard blink.ino and it worked correctly.
so why I can not change led_pin to another pin such as led_pin=22?

MCUdude, 2017-01-20:
of course you can. try uploading this code:
``` c++
uint8_t myPin = 22;

void setup() 
{
  pinMode(myPin, OUTPUT);
}

void loop() 
{
  digitalWrite(myPin, HIGH);
  delay(1000);
  digitalWrite(myPin, LOW);
  delay(1000);
}
```

and connect an LED to pin 22:
[https://github.com/MCUdude/MightyCore#pinout](https://github.com/MCUdude/MightyCore#pinout)

AndroidDesigner, 2017-01-20:
thank you very much
I have another question.
I have a "usb to TTL converter". can I use from it (connect Rx and Tx) to upload my sketches to micro circuit?

MCUdude, 2017-01-20:
You can use any USB to Serial converter that has the DTR pin broken out. When I'm not using mye custom AVR dev board, I use a regular [FTDI adapter](http://www.ebay.com/itm/322042543607). Just follow the [wiring instructions](https://github.com/MCUdude/MightyCore#minimal-setup).

closed 2017-01-21 by MCUdude