MCUdude/MightyCore #47: Problem in HC05 Bluetooth module and Mega32A

AndroidDesigner, 2017-01-24:
hi
I want to build my arduino with ATMega32A.
I can burn bootleader to convert ATMega32A to a Arduino.
and then I tested a simple blink example with my mega32A that worked fine.
NOW, I want to add a bluetooth hc05 module to my breadboard arduino (with ATMega32A).
first:
I tested this byc05 module in an arduino uno and wroks perfectly.
in fact I will send a command ( character '0'  & '1' ) from my app and hc05 receive it and a led is off/on according it.
second:
I used from bobuino wiring and connect this hc05 to Mega32A circuit.
but I dont know why received character in this case is an unreadable character.therefore I can not call if-commands in loop() function that results in my code does not work.
I used a lcd to show the received character by hc05 and saw that the received character is not '0' or '1' (for example is sigma , omega , ...).

I used from the setting in the below in Arduino IDE:
BOARD : ATmega32
BOD : 2.7 volt
pinout : bobuino
Clock : 16 MHz External
Compiler LTO : disabled
Port : my arduino board port (COM 5)
Programmer : Arduino As ISP

Why I receive unreadable character instead of  sent commands ['0' or '1' ]?
please help me...

MCUdude, 2017-01-24:
OK, your information is a bit thin.

first you'll need to post a picture/ schematic diagram of your wiring. Most likely there's something wrong with your wiring. How is the HC05 module connected to the ATmega32? Using UART?
Are you sending serial commands from your PC?

Are you uploading the code using a USB to serial adapter, or using an Arduino as ISP or similar?

If you also post your code I might be able to help you

AndroidDesigner, 2017-01-24:
I think that my wiring is correct (below picture).
1. Rx , Tx hc05 to Tx , Rx micro (pin 14 & pun 15) (Serial pins)
2. I used I2C for lcd (pin 22 & 23 micro).
3. external crystal 16 MHz to Xtal1 & Xtal2 pins.
4. one led to pin 8 (for on/off).
5. I used a divider-resistor 1 K to 2 k in Tx pin of the micro (as suggested in the attached picture)
[image]
.

I send command to the hc05 from an android app.
I send '0' and '1' and used this circuit in an arduino uno and worked fine.

[image]

this is its code:
`
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27, 16, 2);

char incomingByte;
int  led_pin = 8;

void setup() {
  lcd.init();
  lcd.backlight();
  Serial.begin(9600);
  pinMode(led_pin, OUTPUT);
}

void loop() {
  if (Serial.available() > 0) {

    incomingByte = Serial.read();
    Serial.print(incomingByte);

    char c2 = Serial.read();
    lcd.clear();
    lcd.write(c2);

    if (incomingByte == '0') {
      digitalWrite(led_pin, LOW);
    }
    if (incomingByte == '1') {
      digitalWrite(led_pin, HIGH);
    }
  }
}
`

MCUdude, 2017-01-24:
There's a potential issue with your code. If you run Serial.read() twice, you'll be storing the next character in the buffer. Try this code:

``` cpp
#include <LiquidCrystal_I2C.h>

LiquidCrystal_I2C lcd(0x27, 16, 2);

byte incomingByte = 0;
int led_pin = 8;

void setup()
{
  lcd.init();
  lcd.backlight();
  Serial.begin(9600);
  pinMode(led_pin, OUTPUT);
}

void loop() 
{
  if(Serial.available() > 0) 
  {
    // Read incomming character
    incomingByte = Serial.read();

    // Echo back to the PC
    Serial.print(incomingByte);

    // Clear LCD and print character
    lcd.clear();
    lcd.write(incommingByte);

    if (incomingByte == '0') 
      digitalWrite(led_pin, LOW);
    
    if (incomingByte == '1')
      digitalWrite(led_pin, HIGH);
    
  }
}
```

You should also probably add some decoupling capacitors to the ATmega32. If not, the microcontroller tends to sometimes act strange and unreliable

AndroidDesigner, 2017-01-24:
I updated the code. I don't use Serial.read() twice.
please look at it again.
your code is same as me...
I don't have "Echo back to the PC" at all.
please help me...

MCUdude, 2017-01-24:
Since it doen't work like it should, you should check if the received character on the ATmega32 actually is correct. The best way of doing that is to connect the TX pin on the ATmega32 to your PC, and use the serial monitor. You can do this by programing a blank sketch on your Arduino UNO, and connect the TX pin on the ATmega32 to the TX pin of your Arduino UNO. This way you're routing the signal through your Arduino UNO into your PC

AndroidDesigner, 2017-01-24:
also I added two capacitor 22pf to XTAL pins and result is same yet...

AndroidDesigner, 2017-01-24:
I followed your instructions and add the below code in a new sketch to arduino uno:

`void setup() {
  Serial.begin(9600);
}

void loop() {
  if(Serial.available()){
    Serial.print((char)Serial.read());
  }
}`

and open serial monitor. this is it's result:

˜

that is an unreadable character.
also I converted byte to char and again the result is unreadable in the serial monitor.

why Serial.read() is not true from hc05 while this code works fine in an arduino uno?

MCUdude, 2017-01-24:
Try replacing:

``` cpp
Serial.print(incomingByte);
```
with

``` cpp
Serial.print(incomingByte, DEC);
```
I'm wondering what characters that's actually printed. The modified line will print the character as a decimal instead

AndroidDesigner, 2017-01-24:
I added DEC also, but the result is same:
æ
One question:
the baud rate 9600 is true for hc05 ?

AndroidDesigner, 2017-01-24:
can you assemble this circuit same as me and test it yourself, also?

MCUdude, 2017-01-24:
I don't have an HC05 module, but the rest seems OK. Try burning the bootloader with the 1 MHz internal oscillator. This removes another error source

MCUdude, 2017-01-24:
I've been using the ATmega32 for a long time with both low and high speed serial transfers.

Try adding a welcome message like
``` cpp
Serial.println("Hello World!");
```
and see if this text shows up in the serial monitor on the PC

AndroidDesigner, 2017-01-24:
this is its result:

Hello World!
æ˜

yes, shows the welcome message in "setup()" method.
but shows an unreadable character in loop(), yet.

AndroidDesigner, 2017-01-24:
this pic is the result of "lcd.write(incomingByte);"
[image]
I receive some characters such as omega.

MCUdude, 2017-01-24:
I've connected my MightyCore development board to my PC. A development board is so much easier to work with rather than a breadboard setup!

This code works just fine for me. My serial monitor replicates the HC05 bluetooth module. If I write the character 1 in the serial monitor, the LED connected to pin 0 lights up. If I send the character 0, the LED goes dark.

``` cpp

byte incomingByte;
int led_pin = 0;

void setup()
{
  Serial.begin(9600);
  pinMode(led_pin, OUTPUT);
}

void loop() 
{
  if(Serial.available() > 0) 
  {
    // Read incomming character
    incomingByte = Serial.read();

    // Echo back to the PC
    Serial.println(incomingByte);

    if (incomingByte == '0') 
      digitalWrite(led_pin, LOW);
    
    if (incomingByte == '1')
      digitalWrite(led_pin, HIGH);
    
  }
}
```

if you connect the RX pin on the ATmega32 to the RX pin on the Arduino UNO as well, you can test this. If it works (the number 48 is printed when sending a 0, 49 when sending a 1), then theres nothing wrong with the ATmega32 or MightyCore

MCUdude, 2017-01-24:
The reason for all this is because we want to find out what device thats causing all the problems. The baud rate on the HC05 is 9600 baud by default, but it it can be changed by sending a command.

I had some problems a while ago when sending and receiving data over RS485. The way I solved this was to hook up my oscilloscope (Rigol DS1054z) and use the built-in decoder to analyze the data in the transmission line. If you had an oscilloscope with decoding capabilities, problems like this could be solved with ease 😉

AndroidDesigner, 2017-01-24:
I tested the below code (without hc05).
my breadboard includes atmega32A and external crystal 16 MHz only.
I sent a constant string (for example "a") from Tx pin of mega32A to Rx pin of arduino uno and opened serial monitor in the uno.
BUT I didn't receive "a" in the serial monitor (and received unreadable characters!).
why?
I think that problem is not related to hc05. because without it, we has unreadable characters yet!

//======================
//Breadboard code:
//======================
`
void setup()
{
  Serial.begin(9600);
}

void loop() 
{
  Serial.print("a");
    delay(1000);
}`

//======================
//Arduino -uno code:
//======================
void setup() {
  Serial.begin(9600);
  
}

void loop() {
  if(Serial.available()){
    Serial.print((char)Serial.read());
    delay(500);
  }
}

please test it...

MCUdude, 2017-01-24:
This is my setup with the code you posted above. The code worked perfectly, and I received the character **a** every half a second.

[image]

MCUdude, 2017-01-24:
I'm almost 100% sure it is something with the physical setup. You don't have 100 nF decoupling capacitors close to the VCC ang GND pins. Have you tried with the internal 1 MHz oscillator? If there's something wrong with the external oscillator, symptoms like weird gibberish characters are normal. Using the internal 1 MHz oscillator eliminates this issue

AndroidDesigner, 2017-01-24:
please place a 32A on a breadboard.
my circuit is as the below. so why my circuit does not work?:(
[image]

MCUdude, 2017-01-24:
you don't have any 22pF capacitors on the crystal. Without these, it won't oscillate properly.
Connect your microcontroller like [this](https://github.com/MCUdude/MightyCore#minimal-setup), and it should work just fine. **Again; did you try using the 1 MHz oscillator option, and re-burn the bootloader?**

AndroidDesigner, 2017-01-24:
I removed external crystal at all (from breadboard) and upload my code with 1 MHz internall oscillator to atmega32A again.
but I received again unreadable characters such as below :(((((
ÿÿÿÿÿÿÿ

AndroidDesigner, 2017-01-24:
OH MY GOD!!!!
YES!
WHEN I removed external crystal and THEN RE-BURN it, it works fine with 1 MHz internal oscillator.
now, what do I do in next step?

MCUdude, 2017-01-24:
The problem is that you don't have all the necessary external components, such as decoupling capacitors and a pullup resistor on the reset pin. Running it at 1 MHz makes it less vulnerable for errors like this. You should now be able to run the sketch you posted [here](https://github.com/MCUdude/MightyCore/issues/47#issuecomment-274929500). If this works, the HC05 should also work fine

AndroidDesigner, 2017-01-24:
No I have.before I had a 10k resistor to Vcc in reset pin. but does not any effect on the result.
I have 22 uf, 22 pf and 100N capacitor. can I use from these as decoupling capacitor in crystal pins?

MCUdude, 2017-01-24:
I would focus on getting everything up and running before starting to think about using the external crystal. 

I can't see a 100 nF decoupling capacitor between pin 9 and 10 on your breadboard. Neither the pullup resistor on the reset pin. This is components that _always_ has to be there to ensure stability.

But seriously, If you're planning to develop on this platform in the future, I strongly recommend you to buy a proper development board. These money will easily pay itself back by not having to struggle with problems like this.

If you're still going to be using a breadboard in the future, you can't skimp on anything. Not a single capacitor, not a single resistor. I still believe that your physical setup is what caused all this

AndroidDesigner, 2017-01-24:
Ok,I will purchase them tomorrow.
can I ask a question.
One of my friends tells me that 32A or 16A are not suitable at all to work as arduino and we should use 328.
are you agree with him?

MCUdude, 2017-01-24:
No, not as all 😉 
The ATmega32 has more IO pins that the ATmega328, it has the same amounts of flash memory and RAM. The _only_ thing that makes the ATmega328 better is that it has two additional PWM pins, and pin change interrupt on all pins.

If you need pin change interrupts, more IO and 6 PWM pins like the ATmega328 has, you can always just use the ATmega324 instead (or the ATmega164 if compared to an ATmega16).

Tell me, what's his argument for using the ATmega328 instead of the ATmega32?

MCUdude, 2017-04-16:
I'm closing this because this issue is a hardware issue, and not really MightyCore's fault at all.

closed 2017-04-16 by MCUdude