MCUdude/MightyCore #49: LowPower Library compiling error

rickhank2016, 2017-02-18:
I have a Atmega324 on breadboard and I have problem when compile in Arduino IDE.
I have modified this line in LowPower.h file:

	#if defined (__AVR_ATmega328P__) || defined (__AVR_ATmega168__) || defined (__AVR_ATmega324P__) 

but the error remain

this is the error:
/Users/yakamoz/Documents/DOCUMENTI 2016/Arduino/libraries/Low-Power/LowPower.h:144:6: error: #error "Please ensure chosen MCU is either 168, 328P, 32U4, 2560 or 256RFR2."
     #error "Please ensure chosen MCU is either 168, 328P, 32U4, 2560 or 256RFR2."
  
How can I solve and put MVCU in PowerDown ?

MCUdude, 2017-02-18:
What kind of ATmega324 do you have? ATmega324A, ATmega324P or ATmega324PA?
Do you have a link to the library?

rickhank2016, 2017-02-19:
I have ATmega324P, I have discovered problem, was that my IDE take library in another place.
Now is ok

rickhank2016, 2017-02-19:
thank you very much

closed 2017-02-19 by rickhank2016

SkeyPr, 2023-04-17:
> 

Hello I am quite late to this discussion but could you please guide me about how you fixed the problem? I am facing the same problem as you were, "Please ensure chosen MCU is either 88, 168, 168P, 328P, 32U4, 2560 or 256RFR2." and would really appreciate if you could help

thanks