MCUdude/MightyCore #50: RX / TX problem

Alfiszcze, 2017-03-07:
Hi....
I try do make DMX reciver from atmega32A or 16A. 
The code works on 3 different arduinos I have [ Uno, Mega2560 and Nano ]. With MightyCore compiles correctly on 32 and 16...but I cannot make it recive signal through RX pin.
This is the code :
`#include <DMXSerial.h>

const int RL11 = 15;
const int GL0 = 16;
const int GL1 = 17;
const int GL2 = 18;
const int GL3 = 19;

#define RL11DefLev  0
#define GL0DefLev  0
#define GL1DefLev  0
#define GL2DefLev  0
#define GL3DefLev  0
void setup() {

  DMXSerial.init(DMXReceiver);
  pinMode(RL11, OUTPUT);
  pinMode(GL0, OUTPUT);
  pinMode(GL1, OUTPUT);
  pinMode(GL2, OUTPUT);
  pinMode(GL3, OUTPUT);
 }
void loop() {
   
   unsigned long lastPacket = DMXSerial.noDataSince();
  if (lastPacket < 5000) {
   
    analogWrite(RL11, DMXSerial.read(1));
    analogWrite(GL0, DMXSerial.read(2));
    analogWrite(GL1, DMXSerial.read(3));
    analogWrite(GL2, DMXSerial.read(4));
    analogWrite(GL3, DMXSerial.read(5));
       } else {
    analogWrite(RL11, RL11DefLev);
    analogWrite(GL0, GL0DefLev);
    analogWrite(GL1, GL1DefLev);
    analogWrite(GL2, GL2DefLev);
    analogWrite(GL3, GL3DefLev);
}`
I'm using [THIS](http://gotronik.pl/img/modul_atmega_zif40_11.jpg) bredboard and connecting incoming TX signal to PD0 pin - RXD. 
Tryied 1Mhz internal and 8 , 12, 16 Mhz external but no use :( ...added defines to libraty but no luck too :(

Any idea ?? Any help ??

MCUdude, 2017-03-08:
Not weird at all. The DMX library is written for "newer" AVR microcontrollers such as the ATmega168 and 328. The UART register names are different. What you can do is A) follow the [library porting guide](https://github.com/MCUdude/MightyCore/blob/master/Library_porting.md) or B) report the issue over at the DMX library repo page

MCUdude, 2017-03-08:
> Will this DMX should work out of the box [ with library port ] on 644 ??

Most likely, as this (and the 1284p) is the "newer" generation with the same UART register names as the ATmega168/328

MCUdude, 2017-03-08:
When it comes to the library porting, you'll most likely have to paste all the macros inside the DMX library header file in order for the compiler to find them when it's compiling the library

Alfiszcze, 2017-03-08:
Wow :) what a fast answer. OK. I have one 644 will try to write to it and report succes [ I hope ] here

Alfiszcze, 2017-03-08:
I done this porting for the 32A in two ways to test it, because in the .h file there was different way to assign UART register names. First as in your method: define <my chip register name> <main register name> - no effect
second way:  define <main register name> <my chip register name> - no effect too.

MCUdude, 2017-03-18:
Any updates? Did it work on the ATmega644p?

Alfiszcze, 2017-03-18:
NO :( Didn't work. I also find that I can use only three pins from bank A (PA0,1,2) in spite of defined all of bank pins. I gave up coz I have no clue how to resolve this situation.

Alfiszcze, 2017-03-18:
OO.... I tried 644, 644p, 644pa

MCUdude, 2017-04-16:
I'm closing this because it's not really MightyCore's fault, but rather the library. If you want to solve this, contact the library creator instead.

closed 2017-04-16 by MCUdude