MCUdude/MightyCore #51: BOD: Brown Out Detection still active when Disabled

simonthechipmunk, 2017-04-04:
I am using one of the 1284p chips on my bike to control rear/front lights, indicator lights and so on.
The system enables itself automatically by switching in a relay which connects it to a steady 5V battery buffered supply as soon as it sees power from the generator. The tricky part is that the chip is powered by the quite unstable supply from the generator (<0.5s) until the relay is powered on. It basically goes like this:

-start driving
-generator runs and powers up the buck/boost converter (5V)
-1284p gets power via some diodes (3.8V PROBLEM!!). This makes sure only the chip starts up and no other components drain the supply
-1284p uses that voltage to trigger a solid state relay via its output pins
-relay clicks in and connects the battery (5V) to the controller and other components

When I burn the boot loader selecting "BOD:Disabled" I never get a proper boot. The µC just does not start at all on generator power. If I then connect an auxiliary supply of 5V it starts up fine.

I then set the option to "BOD:1.8V" and now it's working again as expected but I really don't need BOD as the chip is running on a stabilized 5V.

The BOD option "Disabled" does not seem to actually disable brown out detection but instead sets it to a voltage above at least 3.8V (4.3V presumably)

MCUdude, 2017-04-04:
The only way to make sure the fuses is correct is to use avrdude to read the fuses. If you enable verbose output in the IDE, you can see the avrdude output when the fuses are sat and bootloader burnt. The extended fuse (efuse) is supposed to be 0xff if BOD is disabled. However, this issue sounds like a decoupling problem. Since the applied voltage is very noisy, a sudden drop may cause the microcontroller to reset. I recommend adding some 100nF caps as close at the power pins as possible, and some larger electrolythics (100uF +) at the same rail. If you have access to an oscilloscope, you should probe the voltage while starting your bike to check how bad the ripple really is. 

I'm on vacation right now, so I'm not able to test the issue on my own development board. However, the fuse bit options in the boards.txt file are correct, so it's most likely caused by horrible ripple 😉

koskee, 2017-04-04:
Thinking out loud a bit here,  but I'm just wondering why you want to disable the BOD when you're running off of a supply which is less than stable?  Would that not be what the BOD is specifically designed for? 

Also, if you look at section 7.2 in the Atmel datasheet for the 1284p, relating to the clock, wdt and Bod, there are other fuses you can set depending on what type your clock source is to modify how quickly the mcu boots. Might find some info in there that can address this problem if you hadn't considered this already. 

Another option might be to add a small battery pack (I'd suggest 1.2v ni-mh possibly because you wouldn't need a cc/cv charger like with li-ion) which could be switched into the mcu on start and would be stable immediately and would be charged by the generator. This could also enable the system to work while sitting at a red light for example, judging by the use of indicators it's possible that you ride in traffic sometimes?

Another suggestion,  investigate the use of diodes with a smaller voltage drop to get a higher voltage beyond them, or otherwise modifying your schematics could help achieve what you want. 

Personally I would suggest leaving the Bod on (especially if it functions properly with it) since I'm pretty sure that is what it is intended to do, is to enable the mcu to overcome power supply abnormalities and function properly.  

Just my $0.02

koskee, 2017-04-05:
Also, what speed are you running at? If it's anything fast like >10-12mhz, then the voltage won't be high enough for the mcu to run at that speed at first. You could slow it down and it may work then.. If you need it to run fast maybe include some code to change the clock prescaler and slow it down until it gets the correct voltage? Not 100% sure how one might implement this properly, but clock vs voltage is something to watch out for as a source of these problems.

simonthechipmunk, 2017-04-05:
@MCUdude 
I did some more testing and I can confirm that the fuses are set correctly:

    BOD:Disabled:
    avrdude: reading input file "0xff"
    avrdude: writing efuse (1 bytes):

    Writing | ################################################## | 100% 0.02s

    avrdude: 1 bytes of efuse written
    avrdude: verifying efuse memory against 0xff:
    avrdude: load data efuse data from input file 0xff:
    avrdude: input file 0xff contains 1 bytes
    avrdude: reading on-chip efuse data:

    Reading | ################################################## | 100% 0.01s

    avrdude: verifying ...
    avrdude: 1 bytes of efuse verified

    BOD:1.8V:
    avrdude: reading input file "0xfe"
    avrdude: writing efuse (1 bytes):

    Writing | ################################################## | 100% 0.02s

    avrdude: 1 bytes of efuse written
    avrdude: verifying efuse memory against 0xfe:
    avrdude: load data efuse data from input file 0xfe:
    avrdude: input file 0xfe contains 1 bytes
    avrdude: reading on-chip efuse data:

    Reading | ################################################## | 100% 0.01s

    avrdude: verifying ...
    avrdude: 1 bytes of efuse verified

And I have to say I am completely at fault here because the data sheet says:

    Speed Grades
    0 - 4MHz @ 1.8 - 5.5V
    0 - 10MHz @ 2.7 - 5.5V
    0 - 20MHz @ 4.5 - 5.5V

I am running it at 16MHz....
I also took some measurements of the exact voltage the µC would start at the corresponding setting (all at 16MHz):

BOD1.8V -> 2.5V (No surprise as this is completely out of spec)
BOD2.7V -> 2.77V (still out of spec but working)
BOD4.3V -> 4.39V
BOD off -> 2.32V

Interestingly enough I still get the strange behavior that I can't get the controller to start when BOD is disabled but found that setting it to 2.7V works best for me as my supply voltage is 3.6V. With BOD disabled it just does not boot at all, no matter how fast I spin the wheel. 1.8V is also giving me an occasional failed boot and also requires a faster wheel spin.
This is also why I write all this down just in case someone else runs into the same problem.

A screenshot of my oscilloscope reading is also attached. The ripple isn't actually as bad as I expected it to be. Thanks for your help :+1: 

@koskee 

perfectly analyzed the problem here. Voltage and clock speed seems to be the cause.

To answer your other questions:

I don't really need BOD as the µC is running on a Li-Ion Power Bank most of the time. The only time it isn't is for 0.1s at startup and all it has to do there is boot, initialize inputs/outputs and switch itself to the 5V rail via the relay so it actually gets 5V even before the setup routine is finished.
The only reason all this came up now is because I switched from https://github.com/maniacbug/mighty-1284p to this bootloader because I needed optiboot for wdt support and compatibility with Arduino IDE 1.8.

The battery pack suggestion is already taken care of by the USB power bank I sacrificed for this. Perfect for this application as I don't need to take care of the charging circuit and even get 5V for charging my phone/GPS or run the lighting at the campsite when I'm on a tour:

    Generator -> rectification and smoothing -> 5V DC/DC Buck/Boost converter -> USB power bank -> 5V rail

The startup power for the µC is taken directly after the DC/DC converter but fed through some diodes. These are actually Schottky diodes already (Vf=0.2) but they are linked in series to prevent back feed to some components (USB-Serial adapter for example). Add the voltage drop of the also present optocoupler and all that's left of my 4.8V is 3.66V for starting the 1284.

Thanks for your thoughts. It is working now with a BOD setting of 2.7V
[image]

closed 2017-04-05 by simonthechipmunk

MCUdude, 2017-04-05:
I'm glad you figured out the issue and learned from it. Also, thanks for properly writing all this down; I'm sore someone will benefit from it in the future!

JensGrabner, 2017-04-10:
For Information only: I have a System with 3.3V and 12MHz. Works fine with BOD1.8V and **not** with BOD2.7V. Timing ??  --  Thanks a lot for your work.