MCUdude/MightyCore #53: DHT11 and DS18B20 problems

Ahmed933, 2017-04-11:
Hi, first of all many many thanks to your awesome blogs :)
i'm using Atmega16 as arduino ,, I could burn bootleader to convert it to arduino successfully and i tried many examples (UART, LCD and serial & parallel shift registers ) and it worked very good. 
Now i want to read temp. & humidity from dht11 or temp. from ds18b20  sensors , regarding to dht11 i'm using  (DHT-sensor-library-master)  library but there's no reading at all , but when i wired it with arduino uno , it worked successfully (i want to say that i wired it correctly 👍 ) ,,, the same thing with DS sensor using (Arduino-Temperature-Control-Library-master) library 
after reading library_porting.md  , i did all steps but there's no reading also and i don't know why !!
please help me ...

MCUdude, 2017-04-11:
Hi!
I'm gone for Easter holiday right now, so I'm not able to test anything right now. What DHT11 library are you using? I can see that the [Adafruit DHT library](https://github.com/adafruit/DHT-sensor-library)  has some debug functionality that [can be enabled](https://github.com/adafruit/DHT-sensor-library/blob/master/DHT.h#L17). Try this first, and see what it outputs. 

Both the Adafruit DHT library and the [OneWire library](https://github.com/PaulStoffregen/OneWire) compiles without errors for the ATmega16, so it shouldn't be necessary to use the library porting guide for these.

Ahmed933, 2017-04-12:
here's my code : 
`#include "DHT.h"

#define DHTPIN 0     // what digital pin we're connected to

#define DHTTYPE DHT11   // DHT 11

// Connect pin 1 (on the left) of the sensor to +5V
// NOTE: If using a board with 3.3V logic like an Arduino Due connect pin 1
DHT dht(DHTPIN, DHTTYPE);

void setup() {
  Serial.begin(9600);
  Serial.println("DHTxx test!");

  dht.begin();
}

void loop() {
  // Wait a few seconds between measurements.
  delay(2000);

  // Reading temperature or humidity takes about 250 milliseconds!
  // Sensor readings may also be up to 2 seconds 'old' (its a very slow sensor)
  float h = dht.readHumidity();
  // Read temperature as Celsius (the default)
  float t = dht.readTemperature();
  // Read temperature as Fahrenheit (isFahrenheit = true)
  float f = dht.readTemperature(true);

  // Check if any reads failed and exit early (to try again).
  if (isnan(h) || isnan(t) || isnan(f)) {
    Serial.println("Failed to read from DHT sensor!");
    return;
  }

  // Compute heat index in Fahrenheit (the default)
  float hif = dht.computeHeatIndex(f, h);
  // Compute heat index in Celsius (isFahreheit = false)
  float hic = dht.computeHeatIndex(t, h, false);

  Serial.print("Humidity: ");
  Serial.print(h);
  Serial.print(" %\t");
  Serial.print("Temperature: ");
  Serial.print(t);
  Serial.print(" *C ");
  Serial.print(f);
  Serial.print(" *F\t");
  Serial.print("Heat index: ");
  Serial.print(hic);
  Serial.print(" *C ");
  Serial.print(hif);
  Serial.println(" *F");
}`
i enabled printing debug msgs , and the output is : 

`DHTxx test!
Max clock cycles: 1000
Timeout waiting for start signal high pulse.
Failed to read from DHT sensor!
Timeout waiting for start signal high pulse.
Failed to read from DHT sensor!
`
notice that i'm using 1 Mhz internal OSC  (because if i use any internal or external one it give me errors), BOD : 2.7 , compiler LTO : disabled , pinout : standard

MCUdude, 2017-04-12:
Hmm. Reading the sensor doesn't use any hardware specific instructions, it's just digitalRead and digitalWrite. Are you using the standard pinout, or the Bobuino pinout?

If you have an oscilloscope in reach, you could use this to determine if there's any communication on the line

Ahmed933, 2017-04-15:
i made my own functions to read from DHT11 to find out where is the problem , and here's my code :
`#define DHT11_PIN 2
uint8_t c = 0, I_RH, D_RH, I_Temp, D_Temp, CheckSum;

/* send start pulse or request */
void Request()         
{
  pinMode(DHT11_PIN, OUTPUT);
  digitalWrite(DHT11_PIN, LOW);     /* set to low pin */
  delay(20);                        /* wait for 20ms */
  digitalWrite(DHT11_PIN, HIGH);    /* set to high pin */
  //Serial.println("Done Request"); /* just for debugging */
} 

/* receive response from DHT11 */
void Response()          
{
  pinMode(DHT11_PIN, INPUT);
  while (digitalRead(DHT11_PIN));
  while (digitalRead(DHT11_PIN) == 0);
  while (digitalRead(DHT11_PIN));
  //Serial.println("Done Response");     /* just for debugging */
}

/* receive byte of data from DHT11 */
uint8_t Receive_data()     
{
  int q;
  for ( q = 0; q < 8; q++)
  {
    while (digitalRead(DHT11_PIN) == 0); /* check received bit 0 or 1 */
   // Serial.println("Done LOW");
    delayMicroseconds(1);
    //Serial.println("Done Us");
    if (digitalRead(DHT11_PIN))    /* if high pulse is greater than 30ms */
      c = (c << 1) | (0x01);       /* then its logic HIGH */
    else                           /* otherwise its logic LOW */
      c = (c << 1);
    while (digitalRead(DHT11_PIN));  /// << code stuck here sometimes 
    //Serial.println("Done HIGH");
  }
  //Serial.println("Done bytes"); /* just for debugging */
  return c;
}

void setup() {
  // put your setup code here, to run once:
  Serial.begin(9600);
  Serial.println("DHT11 TEST");
}

void loop() {
  // put your main code here, to run repeatedly:
  Request();
  Response();
  I_RH = Receive_data();               /* store first eight bit in I_RH */
  D_RH = Receive_data();               /* store next eight bit in D_RH */
  I_Temp = Receive_data();               /* store next eight bit in I_Temp */
  D_Temp = Receive_data();               /* store next eight bit in D_Temp */
  CheckSum = Receive_data();             /* store next eight bit in CheckSum */
  
  Serial.print( I_RH , HEX);
  Serial.print("  ") ;
  Serial.print( D_RH , HEX);
  Serial.print("--") ;
  Serial.print( I_Temp, HEX);
  Serial.print("  ") ;
  Serial.print( D_Temp, HEX);
  Serial.print("--") ;
  Serial.print( CheckSum, HEX);
  Serial.println(" ") ;
  delay(2000);
}`
and i tried this code with UNO and it worked very good , but when i try it with the ATMEGA , didn't work !!
And by showing debugging msgs on serial monitor , i finally know where is the problem , the problem is i'm using ATMEGA with internal 1Mhz OSC because as i told u before when i try another OSC it gives me EERORS like this when i try internal 8MHz : 
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x6d
.
.
and this when i try external 8Mhz : 
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x9b
.
.

So since i'm using 1Mhz , DHT11 is faster than the ATMEGA and it couldn't pick up all received bytes and stuck with those while lines in Receive_data() function and the same problem appears with UNO when i add delay or even one syntax (like Serial.print() )between receiving bytes.

So if we could program the ATMEGA with higher freq oscillation, it will work. so please help me to solve the problem of other oscillators.
Thanks in advance :)

MCUdude, 2017-04-16:
You can't read the DHT11 and DS18B20  with a clock frequency of 1 MHz. It's simply too slow. You need to use the 8 MHz internal oscillator or greater.

It seems like you have a bootloader issue, since you're getting the `stk500_getsync()` error. What does your circuit looks like? How of you upload code to the ATmega16? Have you remembered to hit **Burn Bootloader** when switching to 8 MHz external oscillator?

> Select your microcontroller in the boards menu, then select the clock frequency. You'll have to hit "Burn bootloader" in order to set the correct fuses and upload the correct bootloader. 
Make sure you connect an ISP programmer, and select the correct one in the "Programmers" menu. For time critical operations an external oscillator is recommended. 

I suspect you haven't followed this procedure.

I've tested this with my own setup, an it work without any problems @ 8 MHz internal

Ahmed933, 2017-04-16:
you r right , this is my fault , i should burn bootloader again when switching to another osc. , i tried it with external 8MHz and it works fine :) many many thanks to u 👍

MCUdude, 2017-04-16:
Great! 👍

closed 2017-04-16 by MCUdude