MCUdude/MightyCore #56: Issue when using Software SPI from SDFat Lib

SrikanthBala, 2017-04-22:
At the outset thank you for providing the MightyCore. I am currently using a MightyCore for 1284P and have successfully tested a SD card using the SdFat library on the 1284P running at 3.3V on a external 8MHz crystal on a breadboard using hardware SPI.
However when I used the code provided by Greiman for SoftwareSPI (provided actually for 2560) my code first does not compile at all. and gives an error as below :
In file included from F:\arduino-1.8.2\portable\sketchbook\libraries\SdFat\src/SdSpiCard/DigitalPin.h:45:0,
from F:\arduino-1.8.2\portable\sketchbook\libraries\SdFat\src/SdSpiCard/SoftSPI.h:31,
             from F:\arduino-1.8.2\portable\sketchbook\libraries\SdFat\src/SdSpiCard/SdSpi.h:235,

             from F:\arduino-1.8.2\portable\sketchbook\libraries\SdFat\src/SdSpiCard/SdSpiCard.h:29,

             from F:\arduino-1.8.2\portable\sketchbook\libraries\SdFat\src/SdFat.h:27,

             from F:\arduino-1.8.2\portable\sketchbook\Series2_Tx\DumpInfo\DumpInfo\SDSPI1284\SDSPI1284.ino:7:
F:\arduino-1.8.2\portable\sketchbook\libraries\SdFat\src/SdSpiCard/boards/GpioPinMap.h:40:2: error: #error Undefined variant 1284, 644, 324
#error Undefined variant 1284, 644, 324
I have not got this error when using other libraries on 1284P.
Finally I got the code to compile without errors by commenting out everything in GpioPinMap.h and adding the line #include "BobuinoGpioPinMap.h" as I am using the Bobuino pinout. After uploading, I realized my card is not recognized.
I need to use SoftwareSPI as my RC522 with the SD card if both are attached to the same SPI bus.
Kindly provide me with help as to why it does not compile in the first place without modifying Gpio header file and also why it doesnt recognize my card
I am using a SD fat library which says #define SD_FAT_VERSION 20160719
The Mighty Core is also recent I believe as I am using the IDE version 1.8.2 and downloaded it using Boards Manager. I am using Mighty Core MCUdude version 1.0.8
Please help. My code is the same as on this site with pins modified as below :
// An example of the SdFatSoftSpi template class.
// This example is for an Adafruit Data Logging Shield on a Mega.
// Software SPI is required on Mega since this shield connects to pins 10-13.
// This example will also run on an Uno and other boards using software SPI.
//
#include "SdFat.h"
#include <SPI.h>
#if SD_SPI_CONFIGURATION >= 3 // Must be set in SdFat/SdFatConfig.h
//
// Pin numbers in templates must be constants.
const uint8_t SOFT_MISO_PIN = 6;
const uint8_t SOFT_MOSI_PIN = 5;
const uint8_t SOFT_SCK_PIN = 7;
//
// Chip select may be constant or RAM variable.
const uint8_t SD_CHIP_SELECT_PIN = 10;
// SdFat software SPI template
SdFatSoftSpi<SOFT_MISO_PIN, SOFT_MOSI_PIN, SOFT_SCK_PIN> sd;
// Test file.
SdFile file;
void setup() {
Serial.begin(9600);
// Wait for USB Serial
while (!Serial) {
SysCall::yield();
}
Serial.println("Type any character to start");
while (!Serial.available()) {
SysCall::yield();
}
if (!sd.begin(SD_CHIP_SELECT_PIN)) {
sd.initErrorHalt();
}
if (!file.open("SoftSPI.txt", O_CREAT | O_RDWR)) {
sd.errorHalt(F("open failed"));
}
file.println(F("This line was printed using software SPI."));
file.rewind();
while (file.available()) {
Serial.write(file.read());
}
file.close();
Serial.println(F("Done."));
}
//------------------------------------------------------------------------------
void loop() {}
#else // SD_SPI_CONFIGURATION >= 3
#error SD_SPI_CONFIGURATION must be set to 3 in SdFat/SdFatConfig.h
#endif //SD_SPI_CONFIGURATION >= 3
The error I get is :
Type any character to start
Can't access SD card. Do not reformat.
No card, wrong chip select pin, or SPI problem?
SD errorCode: 0X1,0X0

I understand that the latter part may be an issue with the SoftwareSPI or SDfat library and have posted the same at Greiman Gihub page too.

Please help

SrikanthBala, 2017-04-25:
Finally the problem was solved :

First to solve the compiling problem. I needed to modify the Gpio.h file. The original GpioPinMap.h file in the SDfat library under "SdFat/SdFat/src/SdSpiCard/boards/GpioPinMap.h" has the following two lines which need to be changed to compile under the MCUDUDE MightyCore :
Near the end of the file :
#elif defined(ARDUINO_1284P_BOBUINO)
needs to be changed to #elif defined(BOBUINO_PINOUT)
and
#elif defined(ARDUINO_1284P_STANDARD)
needs to be changed to #elif defined(STANDARD_PINOUT)

After this the code compiled and uploaded but no luck. SD card could not be read.
After nearly 24 hrs of breaking my head on multiple SD card adapters and finally fixing one 'impossible to see' internally broken jumper wire, and trying a zillion other combinations, I found in a 2012 Arduino forum that SdFat SoftwareSpi assumes a Standard or Sanguino pinout and not Bobuino pinout. So I switched to Standard pinout names in the code and BINGO ! it worked !
In fact it does not matter whether I select Bobuino board or Standard board under the boards selection for the Mighty Core. It still works on the Standard Pin names only as if the Standard Pins were somehow hardcoded into the SDFat SoftwareSPI code ignoring the definitions in Gpio.h. I dont currently know how to modify SDFat to use the right pins so I'll leave it at that, as I know how to get the the code to work.

Thanks Greiman for the SdFat library and Mcudude for the Mighty Core. 

Incidentally I tried it at both 16MHz and 8Mhz too and it works GREAT !

I hope this helps someone else struggling on similar lines. Bye

closed 2017-04-25 by SrikanthBala

MCUdude, 2017-04-25:
Great to gear you figured it out. If you want to help out others, you can always create a pull request to the library repo with this fix 👍

per1234, 2017-04-25:
I'm working on fixing the incorrect Bobuino pin mapping now but I wasn't planning to submit one to add support for the Mightcore variant definitions so go ahead and do that one @SrikanthBala. Note that apparently all development work is done in a separate repository so I believe that is where the pull requests should be submitted to:
https://github.com/greiman/SdFat-beta
Reference: https://github.com/greiman/SdFat/issues/8

SrikanthBala, 2017-04-26:
per1234, thanks for correcting the pinout. I see you have merged this request with greiman @SdFat. Actually i am pretty new to this forum and am not sure what a pull request is and how it is to be submitted 
BTW, later I realized that if  we changed GpioPinMAp.h as suggested, someone would lose compatibility with the other Mighty cores like say maniacbug. It would be safer t just add these lines and leave the rest as is retaining compatibility with other mighty cores as well

#elif defined(BOBUINO_PINOUT)
#include "BobuinoGpioPinMap.h"
#elif defined(STANDARD_PINOUT)
#include "Standard1284GpioPinMap.h"

Thanks again

PS. I have to yet read the GitHub guide. I'll get on it so that I use it appropriately without getting on someones nerves

per1234, 2017-04-26:
Yes, I fixed the SdFat to work with Bobuino pinout here:
https://github.com/greiman/SdFat-beta/pull/17
The problem was the pin mapping was wrong, they had just duplicated the Standard pinout in BobuinoGpioPinMap.h.

It sounds like it will be a while before this fix makes it into the library but if you want to go back to using Bobuino pinout you can just replace your SdFat/src/SpiDriver/boards/BobuinoGpioPinMap.h file with
https://raw.githubusercontent.com/per1234/SdFat-beta/46c2f2feb1a08c6ff1bf8b39368b7ff91d9d50cd/SdFat/src/SpiDriver/boards/BobuinoGpioPinMap.h.

The cleanest solution to add MightyCore compatibility to the library without losing compatibility with Mighty 1284P is to change [GpioPinMap.h lines 33-38](https://github.com/greiman/SdFat-beta/blob/master/SdFat/src/SpiDriver/boards/GpioPinMap.h#L33-L38) from:
```c++
#elif defined(ARDUINO_1284P_BOBUINO)
#include "BobuinoGpioPinMap.h"
#elif defined(ARDUINO_1284P_SLEEPINGBEAUTY)
#include "SleepingBeautyGpioPinMap.h"
#elif defined(ARDUINO_1284P_STANDARD)
#include "Standard1284GpioPinMap.h"
```
to:
```c++
#elif defined(ARDUINO_1284P_BOBUINO) || defined(BOBUINO_PINOUT)
#include "BobuinoGpioPinMap.h"
#elif defined(ARDUINO_1284P_SLEEPINGBEAUTY)
#include "SleepingBeautyGpioPinMap.h"
#elif defined(ARDUINO_1284P_STANDARD) || defined(STANDARD_PINOUT)
#include "Standard1284GpioPinMap.h"
```

I think you will find it worthwhile to learn a little about using Git and GitHub. It's a bit confusing at first but it's such an amazing tool for contributing and collaborating on open source projects. Once you become somewhat familiar with how it works it only takes a few minutes to submit a fix for a bug and save countless others from the headaches you encountered. It's really a great feeling when you get the email that your pull request was merged.

SrikanthBala, 2017-04-27:
Thanks per1234. Will do the much needed learning.

I had thought about it but was not sure you can || and && in #elif statements. Hence just added 2 new elifs.  I have switched to Standard pinout for now and will use your fix if I use Bobuino again. 

Regards.