MCUdude/MightyCore #57: Getting duplicate files inside submodules

MCUdude, 2017-04-23:
It just hit me; why haven't I used submodules with any of my cores? MightyCore, MegaCore, MiniCore and MajorCore shares the same bootloader source, Arduino core files and libraries. It would me much easier for me to maintain all this if I just had one library repo, one Arduino core files repo and one bootloader repo.

This would be a lot of work, and I'll start with the bootloader first, as I've all ready started to "unify" this.

per1234, 2017-04-24:
One thing to consider is that submodules are not included in the "Download ZIP" file so this will require people to do git clone if they want to install the tip version instead of a release. That probably wouldn't be a huge problem since most of the people who would want to beta test will be able to figure that out easily if you supply some instructions but I could see someone reporting an issue but then not being willing to jump through the hoops to download a git client just to clone and test the fix.

Submodules are also kind of a pain to work with. You might also consider the alternative of subtrees. I do think moving shared files to their own repositories would be a very good thing.

MCUdude, 2017-04-24:
Thanks for the heads up! I didn't know submodules wasn't included in the ZIP download. Definitly a deal breaker. How is subtrees different from submodules? I'm new to all this, but I'd like to clean up all my cores as much as possible for easy maintainance.

(PS: you got a PM over at the Arduino forum 😉)

MCUdude, 2017-04-24:
A submodule is a symbolic link, a subtree is a copy. After a bit of struggling I was able to add the [optiboot_flash repo to MegaCore](https://github.com/MCUdude/MegaCore/tree/avr-100-pin/avr/bootloaders) using a subtree.

MCUdude, 2018-01-16:
I think we're ready to implement Travis CI to MightyCore very soon. Just pushed some quite large commits to the master branch, and it would be great to have the same coverage that MegaCore has 🙂 Need to dig up the guide you wrote me a year ago

per1234, 2018-01-16:
Great news! That thread ended up being pretty rambling so I've put it all together in one place here:
1. Go to https://travis-ci.org
1. Click the "+" here:
[image]
1. Then enable Travis CI for MightyCore by clicking the left side of the switch next to MightyCore on the repository list (image is for MegaCore):
[image]
[image]
1. Click the gear icon to the right of the on/off switch for MightyCore
1. Turn on the option "Build only if .travis.yml is present". That just prevents pointless attempts by Travis CI to build before the configuration file is added.
1. In the Environment Variables section enter `REPORT_GITHUB_TOKEN` in the Name field and the token in the Value field. You can use the same token as you have for MegaCore and MCUdude_corefiles or create a new one following the first part of [these instructions](https://github.com/per1234/arduino-ci-script#creating-a-github-personal-access-token). Make sure the Display value in build log switch is in the off position. That token is used to publish the reports to your CI-reports repository.
1. Click the Add button.

After that, I'm happy to submit a PR to add the .travis.yml file and also one to add MightyCore testing to the MCUdude_corefiles repository if you like.

MCUdude, 2018-01-16:
Great! I've followed the procedure for MightyCore, so it should be ready now!
It would be great it the MCUdude_corefiles was tested against MightyCore as well

per1234, 2018-01-17:
I think I have it all set up:
https://github.com/per1234/MightyCore/tree/travis-ci

Unfortunately Travis CI free builds are currently down:
https://www.traviscistatus.com/incidents/c1s4dlyxd4lf

I'll submit the PR as soon as I get a finished build.

MCUdude, 2018-01-24:
Any updates on this Per? 🙂

per1234, 2018-01-24:
Sorry for the delay. I was struggling to create a build configuration that didn't exceed the maximum job duration, then when I split it into more jobs I hit the maximum job number limit. Then when I pared down the number of jobs I encountered a limitation of the script. So I had to do some work on the script and while I was at it decided to do another high priority fix to include in the new release.

I think I have it all working well now but for some reason Travis CI got stuck on a build that just ran forever and it wouldn't let me stop the build or start a new one. So I haven't been able to get a complete test build of MightyCore using the new script version yet. I decided I needed to just let the thing sit for a while until whatever problem Travis CI was having was dealt with on their end and now it looks like that's happened.

So I've started what I hope to be a successful build:
https://travis-ci.org/per1234/MightyCore/builds/332719121
and will submit the PR when that is finished.

MCUdude, 2018-01-24:
Thank you! I'll close this issue as soon as the current build turns out to be ok

MCUdude, 2018-01-24:
The latest build has passed. Guess we'll ready for a release pretty soon!

closed 2018-01-24 by MCUdude