MCUdude/MightyCore #62: Not working

scargill, 2017-08-10:
So I'm used to using the Manicbug setup... I tried this.. and like the Manicbug version,  It did not know about INTERNAL for the A/D convertor - this is standard in Arduino...

Thoughts?

scargill, 2017-08-10:
https://www.arduino.cc/en/Reference/AnalogReference

MCUdude, 2017-08-10:
You're right, INTERNAL is not defined for the Atmega164/324/644 /1284. I'll add it in the next release. However you can use INTERNAL1V1 or INTERNAL2V56 instead, which I think is more presice since the MCU has two internal references.

scargill, 2017-08-11:
I have to say I agree it is more precise…  just wasn’t aware of it – maybe just put that up somewhere obvious – hey, thanks for doing this – was using a very old version of Manicbug.  I’ll have to check to see your pin assignments are the same :)

MCUdude, 2017-08-11:
I'm doing this because I think other great AVR devices deserves proper Arduino support 👍 
You should check out my other Arduino cores as well! They have become quite popular too 🙂 

Thans for reporting, and thank you for all the great content you provide on your blog. I've read a lot of your ESP8266 related posts!

closed 2017-08-11 by MCUdude

scargill, 2017-08-11:
Oh I most certainly will look - thanks.