MCUdude/MightyCore #65: Atmega32 bootloader with external crystal not working

Crusher7, 2017-10-24:
ATmega32A bootloader burned successfully but i can not uplod any sketch by UART, I'm using FTDI programmer and I always get no response error , I tried to use different crystals with its bootloader without success .. the only bootlooder which is working the (8MHz internal ) .. I tried to use many solutions but error still occurs

MCUdude, 2017-10-24:
I'm quite sure there's something wrong with your setup. Please post a picture of your actual hardware

Crusher7, 2017-10-26:
I followed your schematic to test the bootloader in breadboard also it was printed in PCB within my project but non of them worked .. but while testing in breadboard I tried to burn bootloader for Atmega328p and it was working well with FTDI programmer .. also as I mentioned previously the bootloader  of 8 MHz worked well but not with the external crystals
[image]

MCUdude, 2017-11-05:
So the FTDI adapter works if you use the internal 8 MHz option, but not with any of the external options?

MCUdude, 2018-02-07:
Did you ever figure this out? You can try to apply a 8 MHz clock signal to the XTAL1 pin in order to emulate a crytal.

It's most likely caused by poor connection or a defective crystal.

closed 2018-03-23 by MCUdude