MCUdude/MightyCore #66: Cannot upload with USBtinyISP

Niko-O, 2017-11-06:
I am using a USBtinyISP (see https://github.com/sparkfun/Tiny-AVR-Programmer ) and Arduino IDE 1.8.3.
When I connect the programmer to an ATtiny85 and select the following options it works fine:
[image]
However, when I connect the programmer to an ATmega16 and I use the follosing options, uploading doesn't work because avrdude tries to use a COM port:
[image]
The exact output is as follows:
```
C:\path to arduino library\arduino-1.8.3\hardware\tools\avr/bin/avrdude -CC:\Users\Niko\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.8/avrdude.conf -v -patmega16 -carduino -PCOM2 -b9600 -D -Uflash:w:C:\Users\Niko\AppData\Local\Temp\arduino_build_107267/sketch_nov06a.ino.hex:i
avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Niko\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.8/avrdude.conf"

         Using Port                    : COM2
         Using Programmer              : arduino
         Overriding Baud Rate          : 38400
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0xda
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0xda
[Snip]
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0xda

avrdude done.  Thank you.

Problem uploading to board.  See http://www.arduino.cc/en/Guide/Troubleshooting#upload for suggestions.
```
I have tried selecting the non-"(MightyCore)" USBtinyISP but same thing.
The COM2 it keeps using is a virtual com port routed to COM3. There is nothing connected to that and I cannot unselect it in Arduino IDE. There are no other COM ports shown in the device manager.

When I change everything back to the ATtiny85 settings but connect the ATMega16 anyways I get more output and other error messages:
```
C:\path to arduino library\arduino-1.8.3\hardware\tools\avr/bin/avrdude -CC:\path to arduino library\arduino-1.8.3\hardware\tools\avr/etc/avrdude.conf -v -pattiny85 -cusbtiny -Uflash:w:C:\Users\Niko\AppData\Local\Temp\arduino_build_107267/sketch_nov06a.ino.hex:i 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\path to arduino library\arduino-1.8.3\hardware\tools\avr/etc/avrdude.conf"

         Using Port                    : usb
         Using Programmer              : usbtiny
avrdude: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
         AVR Part                      : ATtiny85
         Chip Erase delay              : 400000 us
         PAGEL                         : P00
         BS2                           : P00
         RESET disposition             : possible i/o
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    12     4    0 no        512    4      0  4000  4500 0xff 0xff
           flash         65     6    32    0 yes      8192   64    128 30000 30000 0xff 0xff
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : USBtiny
         Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude: programmer operation not supported

avrdude: Using SCK period of 10 usec
avrdude: initialization failed, rc=-1
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

An error occurred while uploading the sketch
```
Which seems somewhat reasonable.
I have tried running the following command, a mixture between the two commands executed above, manually from cmd:
```
"C:\path to arduino library\arduino-1.8.3\hardware\tools\avr/bin/avrdude" -CC:\Users\Niko\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.8/avrdude.conf -v -patmega16 -cusbtiny -Uflash:w:C:\Users\Niko\AppData\Local\Temp\arduino_build_107267/sketch_nov06a.ino.hex:i
```
That gives a slightly different output (e.g. "AVR Part : ATmega16" instead of "AVR Part : ATtiny85") but fails with the same error messages ("programmer operation not supported", "initialization failed, rc=-1", etc.).
I have tested this with both ATmega16 I have. I am confident that neither of them is broken.
I'm at my wit's end. Any help would be appreciated.

MCUdude, 2017-11-06:
Hi!
This is what the correct avrdude command should look like
`C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -C\path\Documents\Arduino\hardware\MightyCore\avr/avrdude.conf -v -patmega16 -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0xd4:m -Ulfuse:w:0b10111111:m`

The USBtinyISP should not appear as a virtual serial port, so that's clearly the issue here:
`..../avrdude.conf -v -patmega16 -carduino -PCOM2 -b9600 -D -Uflash.....`

I assume you didn't choose "Arduino as ISP" as the programmer, so it maybe a bug in the IDE. Do you get the same output with IDE 1.8.5?

Niko-O, 2017-11-06:
This is the output produced by the command you wrote (with adjusted file paths):
```
"C:\path to arduino library\arduino-1.8.3\hardware\tools\avr\bin\avrdude.exe" -CC:\Users\Niko\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.8/avrdude.conf -v -patmega16 -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0xd4:m -Ulfuse:w:0b10111111:m

avrdude.exe: Version 6.3, compiled on Jan 17 2017 at 12:00:53
             Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
             Copyright (c) 2007-2014 Joerg Wunsch

             System wide configuration file is "C:\Users\Niko\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.8/avrdude.conf"

             Using Port                    : usb
             Using Programmer              : usbtiny
avrdude.exe: usbdev_open(): Found USBtinyISP, bus:device: bus-0:\\.\libusb0-0001--0x1781-0x0c9f
             AVR Part                      : ATmega16
             Chip Erase delay              : 9000 us
             PAGEL                         : PD7
             BS2                           : PA0
             RESET disposition             : dedicated
             RETRY pulse                   : SCK
             serial program mode           : yes
             parallel program mode         : yes
             Timeout                       : 200
             StabDelay                     : 100
             CmdexeDelay                   : 25
             SyncLoops                     : 32
             ByteDelay                     : 0
             PollIndex                     : 3
             PollValue                     : 0x53
             Memory Detail                 :

                                      Block Poll               Page                       Polled
               Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
               ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
               eeprom         4    10   128    0 no        512    4      0  9000  9000 0xff 0xff
               flash         33     6   128    0 yes     16384  128    128  4500  4500 0xff 0xff
               lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
               efuse          0     0     0    0 no          0    0      0     0     0 0x00 0x00
               signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
               calibration    0     0     0    0 no          4    0      0     0     0 0x00 0x00

             Programmer Type : USBtiny
             Description     : USBtiny simple USB programmer, http://www.ladyada.net/make/usbtinyisp/
avrdude.exe: programmer operation not supported

avrdude.exe: Using SCK period of 10 usec
avrdude.exe: initialization failed, rc=-1
             Double check connections and try again, or use -F to override
             this check.

avrdude.exe done.  Thank you.
```
Note that I have used `C:\Users\Niko\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.8/avrdude.conf` and not `C:\Users\Niko\Documents\Arduino\hardware\MightyCore\avr/avrdude.conf` as your command would have suggested. There is no `MightyCore` folder in `C:\Users\Niko\Documents\Arduino\hardware\`, only `attiny`. That may be due to the fact that I use the portable version of Arduino IDE and not the installed one.
"Using Port : usb" looks good, but it doesn't work.

>I assume you didn't choose "Arduino as ISP" as the programmer
Correct. I have never used that option before. I have tried it now and it tries to use a COM port too.
Arduino IDE 1.8.5 does the same thing. The settings look the same and the error messages are the same too.
I have tried the above command with 1.8.5 instead of 1.8.3 but the result is the same.

Niko-O, 2017-11-15:
I don't want to come over as impatient but I'm pretty much stuck until I can work this out. Please feel free to suggest any ideas of what I could try or anything which could be of help. Because I have no idea what's wrong.

MCUdude, 2017-11-15:
I'm sorry, but I've never experienced this issue before. I use an USBtinyISP as my main programmer, and it works every time.

I'd start by updating the Arduino AVR boards under boards manager (latest version is 1.6.20). If this doesn't help I'd do a proper installation of the IDE.

Just for reference, please try to install [ATTinyCore](https://github.com/SpenceKonde/ATTinyCore). It's built the same way as MightyCore, so we should get an indication

Niko-O, 2017-11-16:
I have installed Arduino 1.8.5 as a full installation.
According to the board manager there are no updates for "MightyCore by MCUdude version 1.0.8" (it is not listed the "Updatable" list). The portable 1.8.3 and 1.8.5 say the same thing. Removing MightyCore and re-adding it does not change that. BTW: 1.0.8 is the highest version I can select in the board manager.
ATtiny85 still works fine.
ATmega16 still expects a COM port and the command 
`"C:\Program Files (x86)\arduino-1.8.5\hardware\tools\avr\bin\avrdude.exe" -CC:\Users\Niko\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.8/avrdude.conf -v -patmega16 -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0xd4:m -Ulfuse:w:0b10111111:m` 
(same as https://github.com/MCUdude/MightyCore/issues/66#issuecomment-342241051 but with the path adjusted to the non-portable version) gives exactly the same output as before.

I have installed Mighty-1284p  but I am not quite sure which board I should use. None of them seem to apply to my setup. All of those boards try to use a COM port too.

Does the USBtinyISP create a COM port for you?

MCUdude, 2017-11-16:
The USBtinyISP should not create a COM port, it should show up as a native USB device.

When it comes to Mighty-1284p you can just select a board in the list and hit "Burn Bootloader". If it works like it should, you should get an error saying that the attached MCU is not an ATmega1284p

Niko-O, 2017-11-21:
Burning the bootloader with any of the Mighty-1284p boards still tries to use a COM port.
Executing the command `"C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude" "-CC:\Program Files (x86)\Arduino\hardware\tools\avr/etc/avrdude.conf" -v -patmega1284p -cusbtiny -D -Uflash:w:C:\Users\Niko\AppData\Local\Temp\arduino_build_947707/BlinkTest.ino.hex:i`  manually still results in the same "programmer operation not supported" error.

I have installed Arduino IDE on a laptop with Linux on it and tried again.
Programming an ATtiny85 still works as expected. Programming an ATmega16 (I have installed the same boards as before) does again try to use a COM port and when I execute the command `/home/a/Desktop/arduino-1.8.5# ./hardware/tools/avr/bin/avrdude -C/root/.arduino15/packages/MightyCore/hardware/avr/1.0.8/avrdude.conf -v -patmega16 -cusbtiny -e -Ulock:w:0x3f:m -Uefuse:w:{bootloader.extended_fuses}:m -Uhfuse:w:0xd4:m -Ulfuse:w:0b10111111:` manually I still get the same "programmer operation not supported" error.

Well.
By pure luck I saw an Amazon review on an ATmega32 saying that those chips ship with clock source set to an external 1MHz crystal and in order to change that it is necessary to connect such a crystal and then change the clock source. I remembered that the board the chip was originally plugged into had such an external crystal so I decided to plug it back in and jerry-rigged the necessary connections.
And voila. An LED blinks after flashing a simple blink program.
I cannot believe this was such a stupid mistake.

Either way: Arduino IDE trying to use a COM port is still a problem. Any idea what could be causing this?

And I have another weird problem where I can only flash the program once to the ATmega16. If I flash it a second time the program appears to crash somewhere in the middle (e.g. a LED turns on but never turns off), but **only** when I recompile the program inbetween. I can temporarily fix the problem by burning the bootloader and then flashing the program again.
Very weird.

MCUdude, 2017-11-21:
> By pure luck I saw an Amazon review on an ATmega32 saying that those chips ship with clock source set to an external 1MHz crystal and in order to change that it is necessary to connect such a [...]
And voila. An LED blinks after flashing a simple blink program.
I cannot believe this was such a stupid mistake.

By default they ship with the 1 MHz internal oscillator activated. An external oscillator should not be necessary if the chip is brand new and have the default fuses.

First, You should not care about the `programmer operation not supported` error. My own USBtinyISP causes the same error too, but I'm still able to burn the bootloader just fine.

After you've burned the bootloader, an LED attached to Digital pin 0 (PB0) should flash twice every second (The MCU does this when the bootloader does not have any program to run). You've now able to upload your code using a USB to serial adapter; See the schematic in the README for more info. If you still want to use your programmer to upload the code to your ATmega16, you need to hold down the shift-key while hitting the "Upload" button in Arduino IDE.

If you use a USB to serial adapter to upload code to the ATmega16, and it works only once, you most likely have an improper reset circuity. You need to use a 100n capacitor and a 10k pullup resistor  in order for the auto reset to work reliably. A picture of you complete hardware setup would make it easier for me to help you debug your system 😉

Niko-O, 2017-11-22:
Thanks for the insight.
The chip was used in a board I made in school about 5 years ago. Unfortunately I don't have the eagle files anymore, just a few low-resolution screenshots from back then. Here's the layout:
[image]
[image]
The second one is a bit more up-to-date but still not exact. Electrically they both reflect the actual board except for the 10 ohm resistor between the 10-pin-header and the +5V route. Not sure why that was there in the first place but I shorted it out.
The SMD part marked with "A" is a capacitor of which I don't know the capacitance. "B" is a 1kOhm resistor. Not sure how exact those values need to be. The datasheet doesn't appear to say anything about that.
> If you use a USB to serial adapter to upload code to the ATmega16, and it works only once, you most likely have an improper reset circuity. 

I'm not using such an adapter right now. I connect the chip directly to the USBtinyISP.

Holding shift while pressing the upload-button works. Thank you very much for that hint. For now it even fixes the problem about only being able to flash once. Not sure whether that's a coincidence, though.
I do have a USB to serial adapter but I'm not convinced yet that it's worth the effort getting it to work. But I'll keep it in mind.

And for laughs here's a picture of what my setup looks like:
[image]

I ordered a bunch of ATmega32P chips which I'll use in the project I'm trying to build. I'll try to get those to work too.

Thank you very much for your help.

MCUdude, 2017-11-23:
Great to hear!
You should know that uploading code using a USB to serial adapter works very well, and is just as reliable as a dedicated programmer. It's also great for serial debugging 👍

closed 2017-11-23 by MCUdude