MCUdude/MightyCore #67: Serial works with faults

DuyTrandeLion, 2017-11-23:
Hi,

I am using the MightyCore with a 1284P and I ran a Serial test. The serial is not working properly, all I received on the terminal is mess of characters. My USB to Serial is oke because I tested it carefully. I tried to adjust the baud rate but it also didn't work. Is there any cause or solution for this? 
Thank you

MCUdude, 2017-11-23:
So, are you using the internal or the external oscillator in the ATmega1284? Are you able to upload your code using the USB to serial adapter? What does your setup looks like? Do you have a picture? 
Also, please post your bare minimum code to make it easier for me to help you out.

There's many things that can cause issues like this, but it's most certainly a timing issue.

MCUdude, 2018-01-05:
Did you figure this out?

DuyTrandeLion, 2018-01-08:
Sorry for my late reply, I was too busy.
The reason is that I used the wrong frequency of the crystal. After I configured it again, the serial worked smoothly.
Thank you so much for your support

MCUdude, 2018-01-08:
Great! Thanks for using MightyCore 🙂

closed 2018-01-08 by MCUdude