MCUdude/MightyCore #68: Ps2keyboard not working on atmega1284 16mhz

Jarvis785, 2017-11-24:
Please anyone help..  ps2keyboard not working

MCUdude, 2017-11-24:
Really? How on earth should I be able to help you when you've provided _zero_ details?

Jarvis785, 2017-11-24:
Irq pin bobduino digital 2
Data pin A0

It's not receiving any key input

MCUdude, 2017-11-24:
I guess you've connected the keyboard to the wrong inputs on the microcontroller then? Try some other pins

Jarvis785, 2017-11-24:
I've tried different pins .. still not working . Anybody here worked with ps2keyboard on mightycore ?

MCUdude, 2017-12-15:
A quick look at the library source code reveals an easy error; support for ATmega1284p isn't added to the library. You can do so by modifying the [int_pins.h](https://github.com/PaulStoffregen/PS2Keyboard/blob/master/utility/int_pins.h) file.

Anyways I'm closing this issue because it has nothing to do with the MightyCore source code. It's caused by a third party library that I have no influence on. If you're still experiencing issues you should probably create an issue over at the library repo.

closed 2017-12-15 by MCUdude

Jarvis785, 2017-12-15:
Mcudude plese help me out can you plese send me a edited int_pins.h or tell me what lines to add to thw file . Please help me out

drawkula, 2017-12-15:
@Jarvis785 
Please open an issue about this at [https://github.com/PaulStoffregen/PS2Keyboard/issues](https://github.com/PaulStoffregen/PS2Keyboard/issues). That's where changes to that file should be made and documented.