MCUdude/MightyCore #70: Controller Atmega 644p(8MHz) reboot every 200ms

pklg, 2017-12-05:
Hi All.
Now I am designing project with using Atmega 644p (8MYz), Arduino IDE 1.6.8.
After writting of bootloader in an controller is done the Led blinks twice with period one second. After loading os sketch in an controller it teboots every 200ms. It seems that WDT have default setting 200ms.
I tried in setup() configure WDT like **wdt_enable(WDTO_8S);**, but this not helped.
This setup():
`void setup()
{
  wdt_enable(WDTO_8S);
  while (!Serial1); // wait until serial console is open, remove if not tethered to computer
  Serial1.begin(SERIAL_BAUD);

  DEBUGln("Arduino RFM69HCW Transmitter");

  // Hard Reset the RFM module
  pinMode(RFM69_RST, OUTPUT);
  digitalWrite(RFM69_RST, HIGH);
  delay(100);
  digitalWrite(RFM69_RST, LOW);
  delay(100);

  // Initialize radio
  radio.initialize(FREQUENCY, NODEID, NETWORKID);
  if (IS_RFM69HCW)
  {
    radio.setHighPower();    // Only for RFM69HCW & HW!
    Serial1.println("High Power setted");
  }
  radio.setPowerLevel(31); // power output ranges from 0 (5dBm) to 31 (20dBm)
  Serial1.println("Power setted = 31");
  radio.encrypt(ENCRYPTKEY);
  Serial1.println("Radio is encripted");
  /*//---Set low bitrate----
    radio.writeReg(0x03,0x1A);  //4.8kb/s    //(0x03,0x68)RegBitrateMsb 1200 bitrate
    radio.writeReg(0x04,0x0B);  //4.8kb/s    //(0x04,0x2B)RegBitrateLsb 1200 bitrate
    radio.writeReg(0x05,0x00);      //RegFdevMsb     2000
    radio.writeReg(0x06,0x52);      //RegFdevLsb     2000
    radio.writeReg(0x19,0x40|0x10|0x05);      //RegRxBw  DccFreq:010, RxBw_Mant:24, RxBw_Exp:5
    radio.writeReg(0x18,0x00|0x00|0x01);      //RegLna  LnaZin:50ohm, LowPower:Off, CurrentGain:MAX
    //----------------------
  */

  pinMode(LED, OUTPUT);
  DEBUG("\nTransmitting at ");
  DEBUG(FREQUENCY == RF69_433MHZ ? 433 : FREQUENCY == RF69_868MHZ ? 868 : 915);
  DEBUGln(" MHz");

  timer.setInterval(10000, RFM_module);
  wdt_reset();
}`

This is a piece of the serial output:
`
Arduino RFM69HCW Transmitter
High Power setted
Power se
Arduino RFM69HCW Transmitter
High Power setted
Power se
Arduino RFM69HCW Transmitter
High Power setted
Power se

`
What could be causing this behavior?
Thank You.
Leonid.

MCUdude, 2017-12-05:
What happens if you just comment out everything that has to do with the WDT; does it still reset itself? I suspect this might be a power related issue (the radio causes the voltage to drop below 2.7V, and the ATmega644 resets itself). Do you power our ATmega644 from 5V or 3.3V? 

Please add the following line right after Serial1.begin
``` c
DEBUGln(MCUSR);
```

The MCUSR register contains the reset flags. By reading this register you're able to determine what caused the microcontroller to reset itself.

JensGrabner, 2017-12-06:
In a 3V3 System - i use brown-out detection (BOD) = 1V8  not 2V7 -- while .. the µC will not started. You can change the BOD Voltage .. with burning the Bootloader. I hope this is helpfull.

pklg, 2017-12-06:
Thanks to All.
In first I did DEBUGln(MCUSR); in setup and got message MCUSR=6. It means that reset on BOR occurs.
Then I rebooted Bootloader with BOD=1,8V. But after this Led not blinks after Reset and loaded program in µC  not works - ComPort terminal  outputs nothing. 
AVRdude 6.3.
This is message AVRdude after loading of program in µC:
`
Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x1e960a (probably m644p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
         To disable this feature, specify the -D option.
avrdude: erasing chip
avrdude: auto set sck period (because given equals null)
avrdude: warning: cannot set sck period. please check for usbasp firmware update.
avrdude: reading input file "C:\Users\A26A~1\AppData\Local\Temp\build1d780bd2df423b0e69106791470fc8b7.tmp/RFM_Gateway.ino.hex"
avrdude: writing flash (13454 bytes):

Writing | ################################################## | 100% 7.89s

avrdude: 13454 bytes of flash written
avrdude: verifying flash memory against C:\Users\A26A~1\AppData\Local\Temp\build1d780bd2df423b0e69106791470fc8b7.tmp/RFM_Gateway.ino.hex:
avrdude: load data flash data from input file C:\Users\A26A~1\AppData\Local\Temp\build1d780bd2df423b0e69106791470fc8b7.tmp/RFM_Gateway.ino.hex:
avrdude: input file C:\Users\A26A~1\AppData\Local\Temp\build1d780bd2df423b0e69106791470fc8b7.tmp/RFM_Gateway.ino.hex contains 13454 bytes
avrdude: reading on-chip flash data:

Reading | ################################################## | 100% 7.59s

avrdude: verifying ...
avrdude: 13454 bytes of flash verified

avrdude done.  Thank you.

`

MCUdude, 2017-12-06:
No idea why the lde don't blink, but you really need to fix you power supply. The radio draws a low if current, and you need a power supply that can deliver enough + a lot of decoupling capacitors to prevent the voltage to drop like this.

pklg, 2017-12-06:
About power source. I I worked with these program using Anarduino RFM69HW and used the same power sourse. The program worked well.  Anarduino based on an Atmega 328p. Now I use Atmega 644p because I need more flash and SRAM memory.
Now about message of AVRdude:
avrdude: Device signature = 0x1e960a (probably m644p)
**avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
To disable this feature, specify the -D option.
avrdude: erasing chip**
"Erasing chip" - does it means bootloader will be erased as well?

MCUdude, 2017-12-15:
Sorry, totally forgot to reply. If you get this message:
```
avrdude: Device signature = 0x1e960a (probably m644p)
avrdude: NOTE: "flash" memory has been specified, an erase cycle will be performed
To disable this feature, specify the -D option.
avrdude: erasing chip
```

While uploading using the USB to serial adapter, the bootloader does not get erased

pklg, 2017-12-22:
Hi.
Now I cannot to load bootloader on an Atmega 644p. Arduino IDE 1.8.5.
This is what I got:
``
C:\Users\Леонид\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/bin/avrdude -CC:\Users\Леонид\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.8/avrdude.conf -v -patmega644p -cstk500 -PCOM24 -e -Ulock:w:0x3f:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xf7:m 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Леонид\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.8/avrdude.conf"

         Using Port                    : COM24
         Using Programmer              : stk500
         AVR Part                      : ATmega644P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       2048    8      0  9000  9000 0xff 0xff
           flash         33     6   256    0 yes     65536  256    256  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : STK500V2
         Description     : Atmel STK500
         Programmer Model: STK500
         Hardware Version: 15
         Firmware Version Master : 2.10
         Topcard         : Unknown
         Vtarget         : 3.3 V
         SCK period      : 0.5 us
         Varef           : 0.0 V
         Oscillator      : Off

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.00s

avrdude: Device signature = 0x1e960a (probably m644p)
avrdude: erasing chip
avrdude: reading input file "0x3f"
avrdude: writing lock (1 bytes):

C:\Users\Леонид\AppData\Local\Arduino15\packages\arduino\tools\avrdude\6.3.0-arduino9/bin/avrdude -CC:\Users\Леонид\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.8/avrdude.conf -v -patmega644p -cstk500 -PCOM24 -Uflash:w:C:\Users\Леонид\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.8/bootloaders/optiboot_flash/atmega644p/optiboot_flash_atmega644p_115200_16000000L.hex:i -Ulock:w:0x0f:m 
Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x3f:
avrdude: load data lock data from input file 0x3f:
avrdude: input file 0x3f contains 1 bytes
[... 127 lines cut ...]
avrdude: 65536 bytes of flash verified
avrdude: reading input file "0x0f"
avrdude: writing lock (1 bytes):

Writing | ################################################## | 100% 0.02s

avrdude: 1 bytes of lock written
avrdude: verifying lock memory against 0x0f:
avrdude: load data lock data from input file 0x0f:
avrdude: input file 0x0f contains 1 bytes
avrdude: reading on-chip lock data:

Reading | ################################################## | 100% -0.00s

avrdude: verifying ...
avrdude: WARNING: invalid value for unused bits in fuse "lock", should be set to 1 according to datasheet
This behaviour is deprecated and will result in an error in future version
You probably want to use 0xcf instead of 0x0f (double check with your datasheet first).
avrdude: 1 bytes of lock verified

avrdude done.  Thank you.

``
The sketches are loading nice.
What mey be?

pklg, 2017-12-22:
I add to previous post: the loaded sketch not work.
`
void setup() {
  // put your setup code here, to run once:
  while (!Serial1); // wait until serial console is open, remove if not tethered to computer
  Serial1.begin(115200);
  Serial1.println("Setup done");
}

void loop() {
  // put your main code here, to run repeatedly:
  Serial1.println("Loop is working");
  delay(5000);
}
`
On an terminal output nothing.

JensGrabner, 2018-01-02:
Are you sure that you use a 8MHz Frequency  --  while youre Bootloader is .. "optiboot_flash_atmega644p_115200_16000000L.hex" for 16Mhz.

MCUdude, 2018-02-07:
Did you ever solve this issue?

It's most likely caused by your radio. When the radio is initialized the supply voltage to the microcontroller drops. Does the microcontroller work if you physically disconnect the radio module?
If so, it's a hardware issue an nothing to do with MightyCore/the code.

pklg, 2018-02-08:
Hi.Yes, I solved this. Problem was in function Modbus task(), which was included in the main loop of program. Just not all what is needed for work of thisfunction was setted up and initialized. Now all work nice.Leonid.

MCUdude, 2018-02-08:
Great!

closed 2018-02-08 by MCUdude