MCUdude/MightyCore #72: Can't use `Ethernet` from `Arduino/libraries`

codmpm, 2017-12-22:
Hi,

I've just built a prototype with an ATMega644PA for a networking project. As I'm using an WizNet W5200/W5500 I want to use [Paul Stoffregens excelent Ethernet Lib](https://github.com/PaulStoffregen/Ethernet) which runs flawlessly on an ATMega328p.

The problem now is, that `Ethernet` is always used from `Arduino15/packages/MightyCore/hardware/avr/1.0.8/libraries/Ethernet` and not from `/Users/<user>/Documents/Arduino/libraries/Ethernet`.

Any hint's besides deleting `Arduino15/packages/MightyCore/hardware/avr/1.0.8/libraries/Ethernet` to use the user supplied library? 

I'm using Arduino 1.8.5 on OSX High Sierra.

Thank you!

per1234, 2017-12-22:
Normally the libraries in the sketchbook folder (`/Users/<user>/Documents/Arduino/libraries` in your case) have include priority but the Arduino IDE first gives preference to the library with the best architecture match. MightyCore's Ethernet library has this line in library.properties:
```
architectures=avr
```
PaulStoffregen/Ethernet has this:
```
architectures=*
```
The explicit architecture match is considered better than the wildcard match so the MightyCore library is used. If you change line 9 of `/Users/<user>/Documents/Arduino/libraries/Ethernet/library.properties` to:
```
architectures=avr
```
the architecture matches will be the same and thus the library in the sketchbook (PaulStoffregen/Ethernet) will get include priority.

codmpm, 2017-12-22:
Thank you very much for that info, but thats odd. I was sure, that it worked in the past on the mentioned 1284p.

Any chance to control this from the application? 

I'm currently developing a hardware design (CC-BY-SA, https://twitter.com/pregopm/status/943488233081573376) based on an 328p. To expand the project I'll need a second UART and thus the 644P came into play.
This board will have a software preinstalled but should be hackable by anyone. Don't know if hacking the suggested libraries is a good starting point for that. I was thinking of providing info like: Install MightyCore, Install PaulStoffregen/Ethernet, Go!

per1234, 2017-12-22:
>I was sure, that it worked in the past on the mentioned 1284p.

The architecture match based priority inversion (https://github.com/arduino/arduino-builder/pull/123) was added in Arduino IDE 1.6.9/arduino-builder 1.3.15 to address https://github.com/arduino/Arduino/issues/4064. So if you used Arduino IDE 1.6.8 or older then the unmodified PaulStoffregen/Ethernet library had include priority.

codmpm, 2017-12-22:
Ouh... so thats the point. Ok.

Thank you very much for the information. Maybe @PaulStoffregen can say something about this. 
I'm unaware of a good out-of-the-box solution... is forking Paul's library a good idea? Maybe not...

PaulStoffregen, 2017-12-22:
If I were to buy a board for future testing with MightCore, which would be the best to get?  I'm less concerned about the cost and most concerned about saving time testing.

codmpm, 2017-12-22:
Thank you Paul!

I don't know if it's good, but MightyCore suggests their dev board at their tindie store: https://www.tindie.com/products/MCUdude/dip-40-arduino-compatible-development-board/

If you want I can send you some DIP-40 ATMega 1284/644. But I think shiping to America is too expensive in comparison.

PaulStoffregen, 2017-12-22:
That board doesn't have the Wiznet hardware, nor a place to plug in any of the ethernet shields.  Not ideal time-efficient for testing, like I do with so many other boards...

codmpm, 2017-12-22:
You are right... not thought about the WizNet... I will get back to you if I find an 644/1284 with arduino layout.

codmpm, 2017-12-22:
Got it. The Freetronics Goldilocks: https://www.freetronics.com.au/products/goldilocks-arduino-compatible-with-atmega1284p-mcu

PaulStoffregen, 2017-12-22:
It's discontinued.

"This product is currently sold out" on the product page, and this:
http://forum.freetronics.com/viewtopic.php?f=7&t=5707

codmpm, 2017-12-23:
Hmm. Thats unsatisfying. I will look out for another solution and get back to you.

SalvorinFex, 2017-12-24:
Hey Folks, this, I believe is the same board as the Freetronics one:  https://www.tindie.com/products/feilipu/goldilocks-analogue/?pt=ac_prod_search

PaulStoffregen, 2017-12-24:
Can anyone confirm this board really works with MightyCore?  It seems to have its own board package which merely uses a variant of Arduino's core.

MCUdude, 2017-12-24:
Hardware wise the Goldilocks should work with MightyCore. You'll probably need to replace the bootloader, since it's loaded with stk500v2 from the factory.

Another issue is the pinout. The Goldilocks pinout is different from the already established **Standard pinout** and **Bobuino pinout**. 

You can use the [BOBUINO_PINOUT](https://github.com/MCUdude/MightyCore/blob/52b7224713b56ffda9be6b8f03f97b798d28370d/avr/variants/bobuino/pins_arduino.h#L42) and [STANDARD_PINOUT](https://github.com/MCUdude/MightyCore/blob/52b7224713b56ffda9be6b8f03f97b798d28370d/avr/variants/standard/pins_arduino.h#L44) macros to sort out what pinout is currently being used.

PaulStoffregen, 2017-12-24:
I'm not trying to be difficult about this, but if I'm going to buy a board specifically for ongoing testing of my Ethernet fork and several other libs I maintain, it really needs to work "out of the box".  I have a small budget for hardware to test libs, so money's not the problem.  Time is.  I maintain a lot of code, and I do test on many different boards.  I really need test hardware that's ready-to-go with MightyCore in its default configuration.

per1234, 2017-12-24:
I have a list of all the variants used for ATmega1284P and family and consulting it I find the only one with Arduino shield compatible headers and one of the pinouts supported by this core is the Bobuino (or Bobuino2):
http://crossroadsfencing.com/BobuinoRev17/index.html

I'm not aware of any with W5x00 onboard.

I apologize to @MCUdude for mentioning someone else's product here but I do think it would be great if PaulStoffregen wanted to do some testing using MightyCore.

MCUdude, 2017-12-24:
@per1234 No problem mention other products here. My board was designed to be used with the *Standard pinout*, and shield compatibility was never intended. Paul needs a board that works with the Wiznet ethernet shield right out of the box - my board is not the right option 😉 

Merry Christmas everybody! 🎄

codmpm, 2017-12-25:
Thank you everyone for looking into this problem. If I can help, let me know.

codmpm, 2018-03-17:
I'm thinking of building this for an 644/1084 to have a plattform for standard shields, so that @PaulStoffregen can use it without hassle: https://www.hackster.io/temucin/arduino-uno-with-8-times-more-memory-f164b6

Thoughts?

PaulStoffregen, 2018-03-17:
If any Tindie seller offers a known-good board, I'd be happy to order one (well, assuming it's reasonable, like under $100).  I have a small budget for hardware.  Time is the precious commodity in short supply...

codmpm, 2018-03-17:
Thank you guys. 

My goal would be to get MightyCore working with Paul's Ethernet Library so that I can use it for different projects on my side (ATMega 644, 2 UARTs, etc). As these will be/are opensource I try not to let the enduser edit some arduino core files and have it working "out of the box" as much as possible.
The big question here is, is it worth the effort to get this working in comparison to using another platform (ESP8266, ATSAM, etc.)...

Sadly, I currently have not seen an orderable atmega644/Wiznet board to use mightycore with the Teensy EthernetLib.

PaulStoffregen, 2018-03-17:
I believe it should "just work" as long as you have the Wiznet chip connected to the SPI pins and its CS signal to pin 10.  Or you can use CS on any other digitial pins with Ethernet.init(pin) before Ethernet.begin() - the same API that Adafruit added to Ethernet for use of alternate pins.  There's really no reason it should not work on any AVR-based board where the SPI library works.

codmpm, 2018-03-17:
I'm with you. I had the Wiz850io running with an ATMeag644 on a breadboard also with an ESP8266. But the problem that the architectures bite each other an current Arduino platform persists: But on my side I can't get around https://github.com/MCUdude/MightyCore/issues/72#issuecomment-353509025

PaulStoffregen, 2018-03-17:
If Arduino is using the wrong copy of Ethernet, surely you can work around this by moving or deleting the wrong one, so the IDE is forced to only find the copy you intended to use.

PaulStoffregen, 2018-03-17:
And yeah, agreed the laying of the architecture stuff on top of the other complex rules isn't ideal, but that's a decision the Arduino devs made....

codmpm, 2018-03-17:
Absolutely. But this will be in the library folder of Arduino itself. On a Mac I have to explain to people to open the package and edit somehting inside or delete folders there. Not very beginner friendly.

Maybe I'm wrong on this, but on 1.6.8 or older it worked as wanted.

codmpm, 2018-03-17:
Don't get me wrong, for myself this is absolutely ok....

Another idea: On plattform.io I can add the wanted libraries inside the project folder and don't have to use the central ones from the arduino core... I will have a look into this.

PaulStoffregen, 2018-03-17:
Well, we're talking about 2 separate issues here... whether or not my Ethernet fork/redesign work on Mighty, and how the Arduino IDE chooses which library to use.

PaulStoffregen, 2018-03-17:
I'm not going to get sucked into another debate about Arduino's library searching, *especially* now that much of it is coded in Google Go language!

codmpm, 2018-03-17:
You are right!

Your lib works with Mighty without any problems. The problem ist the decision made in the IDE to

PaulStoffregen, 2018-03-17:
Ok then, my work is done here then.

Please, do not tag me again unless there's an issue with code I have published.  I do not wish to get involved in Arduino's library search algorithm.  Take that up with the Arduino devs, not me.  Ok?

codmpm, 2018-03-17:
Sure. And sorry for that :-)
Thank you for commenting on this and making things clear.

PaulStoffregen, 2018-03-17:
Just so my position is clear, while I did contribute to the library search stuff many times in the distant past, that was in the days before Arduino 1.6.6, when it was coded in Java within arduino-core.jar in the IDE.

Since 1.6.6, they moved all that stuff to arduino-builder which is written in Google Go.  I don't use Go, and at least for now, I have no design to start doing so.  They also added a contributor agreement of some sort, which I also have no desire to deal with right now.  I'm **not** planning to get involved in arduino-builder.  You need to deal with the Arduino devs, not me.

codmpm, 2018-03-17:
Is there a possibility to replace `MightyCore/hardware/avr/1.0.8/libraries/Ethernet` with Paul's Ethernet library into Mightycore?

From my point of view, it is far better than the "official" one. The W5100 support is better and you don't need a hack to get W5200/W5500 working.

MCUdude, 2018-03-17:
I don't have any Wiznet hardware, so I'm not able to verify that Paul's library works well with MightyCore compatible chips. I can add it if you can test it properly for me 🙂

codmpm, 2018-03-18:
Sure I'm willing to test as good as I can. Also I can send you a Wiz850io or any other W5x00 board to connect it via SPI to a MightyCore compatible MCU (except ATTiny, though).

Ping me how I can help.

MCUdude, 2018-03-19:
I|d be very greatful if you'd send me some Wiznet hardware! Perhaps I can make them work with MegaCore compatible hardware as well. Send me an email at hansibull {ä} gmail 👍

closed 2018-04-07 by MCUdude

codmpm, 2018-04-07:
WOW! Great. 
Hopefully I have time to test it in the near futere...

commit 8e481193 (2018-04-07): Update Ethernet library
This updated library uses Paul Stoffregen's brilliant ethernet library for Wiznet hardware, and is compatible with MightyCore. Closes #72