MCUdude/MightyCore #73: I am not able to burn bootloader/upload sketch - Atmega 1284p with Arduino IDE 1.8.5.

venkatyaragalla, 2018-01-02:
I am not able to burn bootloader and upload sketch - Atmega 1284p with Arduino IDE 1.8.5.
 - i was able to burn bootloader and upload sketch 6 months back. unfortunately my system was formatted and do the same thing with Arduino IDE 1.8.5 and MightyCore v1.0.8. but I am not able to burn bootloader and upload sketch. please help me to burn bootloader and upload sketch into Atmega 1284p.

 Clock-16 MHz external oscillator
 BOD-2.7v
 LTO -Disabled
 Pinout-Standard
 Variant-1284p
Programmer-Arduino As ISP(MightyCore)

**Minimal setup-**
[Minimal setup ](https://camo.githubusercontent.com/16d2431f552f1fcbeb27c381d0ef9e16289847e3/687474703a2f2f692e696d6775722e636f6d2f7366655444735a2e706e67)

**Error**

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\ursve\Documents\Arduino\hardware\MightyCore\avr/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03

MCUdude, 2018-01-02:
Please enable Verbose output in the IDE settings, and paste the output when you hit "Burn Bootloader"

venkatyaragalla, 2018-01-02:
Thanks for quick reply. Here is output

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\ursve\Documents\Arduino\hardware\MightyCore\avr/avrdude.conf -v -patmega1284p -cstk500v1 -PCOM4 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xf7:m 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\ursve\Documents\Arduino\hardware\MightyCore\avr/avrdude.conf"

         Using Port                    : COM4
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x03
avrdude: stk500_recv(): programmer is not responding
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x03

MCUdude, 2018-01-02:
Ah, it's your programmer; it's not responding to the commands sent from the computer. Not related to MightyCore at all. I don't think you'll be able to burn bootloader to any other chip with that programmer. I suggest buying a proper, dedicated programmer, such as the USBasp or USBtinyISP.

venkatyaragalla, 2018-01-02:
Thank you.I tried Arduino Uno as programmer( i was able to do it previously).

closed 2018-01-03 by venkatyaragalla