MCUdude/MightyCore #75: Not able to Serial.print

FedorAdarichev, 2018-01-10:
I have successfully uploaded a MightyCore bootloader and a sketch to my 1284p clone board.  When I try to use Serial.print to debug, nothing shows up on the serial monitor?  I have Serial.begin(9600); in my setup code, and I chose 9600 as my baud rate on the monitor.  Is this something to do with the bootloader, as I have asked on the Arduino forum and it sounds like I have everything set up correctly from the Arduino IDE and sketch side of things?

epccs, 2018-01-10:
For reference
https://forum.arduino.cc/index.php?topic=521308.30

FedorAdarichev, 2018-01-10:
Our friends at Arduino have figured it out, link is above.

I'ts not a bootloader issue, just me forgetting to put the "1" in Serial1.print(...);

closed 2018-01-10 by FedorAdarichev

epccs, 2018-01-10:
Just an FYI for Hans: your work on MightyCore and the others is very much appreciated. Thank You