MCUdude/MightyCore #84: Consider supporting Sanguino pinout

SpenceKonde, 2018-01-29:
There exist boards with pins numbered with yet another pinout. 

See pert's comment here: http://forum.arduino.cc/index.php?topic=525636

MCUdude, 2018-01-29:
Yeah, the Sanguino pinout... I don't see any reason why anybody would use that pinout except for legacy support. It's a simple fix; I'll consider adding it 👍

per1234, 2018-01-29:
I'm happy to submit a PR for this, including updates to .travis.yml if you decide you want it. I'd like to see MightyCore offer the widest range of support so there's no reason anyone would need to need to use any other hardware package.

standard, bobuino, and avr-developers (the variant used by sanguino) are the three provided by the original maniacbug/mighty-1284p package.  The JChristensen/mighty-1284p fork of that package later added the [sleepingbeauty ](https://github.com/JChristensen/mighty-1284p/blob/v1.6.3/avr/variants/sleepingbeauty/pins_arduino.h)variant. It's quite annoying how many different variants have been created for the ATmega1284/644 so I definitely understand why you would want to limit it to only the "best" ones. I actually have a list of 14 different ATmega1284/644 variants! It's really a lot of work trying to support all of them, which I have attempted to do in a couple of projects.

MCUdude, 2018-01-29:
A PR would be nice! I don't know the history around the avr-developers/Sanguino pinout, but I think users would recognize "Sanguino pinout" rather than "avr-developers pinout".

closed 2018-01-31 by MCUdude