MCUdude/MightyCore #86: Enhancement request: Add 2Mhz and 4Mhz internal OSC

danyhm, 2018-02-20:
Hello,

I've found your great repo for using Arduino with Internal Oscillators due to power saving needs,
However, the code I have needs more than 1Mhz of the clock and could use less than 8Mhz of the clock to save more power since I'll be running below 3.3 Volt and lower voltages need lower MCU frequencies.

I've read the datasheet and it's possible to have 1,2,4 and 8Mhz of internal Oscillator but your project supports 1 and 8 Mhz only.

I suppose It's only a matter of adding some lines of code to the boards.txt file but since I'm not that familiar with the repo yet so I didn't create a pull request.

Can you please consider adding this option? (mini and micro cores could use the same)

MCUdude, 2018-02-20:
Hi!
By configuring the fuse bits (which is the way MightyCore sets the clock speed) you can only set a clock frequency of 8 MHz or 1 MHz internal oscillator. If you want to divide this down you'll have to modify the CLKPR register. MightyCore does not use this register, and I don't plan to do it either. What you can do is to simply modify the boards.txt file and change the f_cpu parameter, and then change the CLKPR register inside your setup() routine.

closed 2018-02-20 by MCUdude

danyhm, 2018-02-20:
[image]
In the datasheet it says you can program the CKSEL to get all 4 frequencies. Am i missing something?

MCUdude, 2018-02-20:
It's true, but is has to be done _within_ the application. When selecting the 1 MHz option in the boards menu, you're actually setting the CKDIV8 fuse (which again controls the CKSEL register directly).

I would have to add some code to the main repo in order to take advantage of the CKSEL fuse. Very few are using 2 and 4 MHz, so I don't think I'm going to add it. But hey, you're free to fork this repo and do as many changes as you like! 😉

danyhm, 2018-02-20:
I'll try to fork and update that part but i don't have the source code for the bootloader to modify. Where can i find it?
Also the CKDIV8 fusebit is not available in Atmega32 ,Atmega16 and atmega8 so i believe you made a mistake here. No hard feelings!
in conclusion for these 3 atmegas (i don't know about others) it's possible to add the other frequencies without touching the application.
I'll try to add and make a pull request.

MCUdude, 2018-02-21:
You can find the the source code for the bootloader [here](https://github.com/MCUdude/optiboot_flash).

You're correct about the ATmega8, ATmega16 and ATmega32. They don't have a separate CKDIV8 fuse, which makes it possible to set frequencies like 2 MHz and 4 MHz. Seems like Atmel did some changes in the next generation with ATmega164/324/644/1284, because they only support 1 MHz an 8 MHz by the [fuse settings](https://github.com/MCUdude/optiboot_flash).

> in conclusion for these 3 atmegas (i don't know about others) it's possible to add the other frequencies without touching the application.

That's correct. I think this also is true for ATmega8515, ATmega8535, ATmega64 and ATmega128 as well. I'll try to add 2 MHz and 4 MHz to the bootloader makefile tonight. This way it's easier if someone wants to run their application at 2 or 4 MHz and want to use a bootloader.

danyhm, 2018-02-21:
That'd be great thank you very much. I looked at the bootloader codes and it'll take a lot of time if I wanted to catch up with what you did and add the new frequencies. Is it possible for you to upload the compiled bootloader as well?

MCUdude, 2018-02-21:
Pre-compiled hex files for 4MHz and 2 MHz can now be found on the [Optiboot_flash repo](https://github.com/MCUdude/optiboot_flash)!

danyhm, 2018-02-21:
Very good, thank you very much, I'll start developing and add PR if i get anything done

MCUdude, 2018-02-21:
>  I'll start developing and add PR if i get anything done

You don't actually need to do that, as 4 MHz and 2 MHz won't be added to any of these cores any time soon. It's not that it requires work, it's just that I don't want to add a bunch of clock frequencies that's seldom used in the main repo. The idea only have the most essential clock frequencies available, while there still exist bootloaders for many other frequencies. If anybody need to use a weird frequency, it's fairly easy to modify the boards.txt to fit their needs and the bootloader already exist. 

The plan is to offer boot loaders for all clock found in the [Wormfood baud rate calculator](http://wormfood.net/avrbaudcalc.php). That should cover most applications for everyone. Still it takes a lot of time to add all this to the **makeall** file, since I'm building for 41 different microcontroller variants.

I'm not trying to be rude here, it's just that I often get request/PRs by users who have done modifications to fit a certain application, and want their code to be added to the main repo afterwards.
Sometimes I found the PR to be useful to the majority, sometimes not.

I hope you understand 🙂

danyhm, 2018-02-21:
No hard feelings, I understand.