MCUdude/MightyCore #90: DTR is wrong

njlinjie, 2018-04-09:
I already  add a 100uf Capacitance betwine DTR and RST,But arduino response : avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x4e, If I Manual click reset switch ,that is ok,why?

per1234, 2018-04-09:
As shown in the schematics, you're supposed to use a 100 **n**F capacitor, not 100 uF.

njlinjie, 2018-04-09:
sorry，that is 104 Capacitance 。I type  wrong.
16.menu.clock.1MHz_internal=1 MHz internal
16.menu.clock.1MHz_internal.upload.speed=9600
16.menu.clock.1MHz_internal.bootloader.sut_cksel_bits=100001
16.menu.clock.1MHz_internal.bootloader.high_fuses=0xd4
16.menu.clock.1MHz_internal.bootloader.file=optiboot_flash/{build.mcu}/optiboot_flash_{build.mcu}_{upload.speed}_{build.f_cpu}.hex
16.menu.clock.1MHz_internal.build.f_cpu=1000000L

njlinjie, 2018-04-09:
USB to ser chip is  cp2102。

MCUdude, 2018-04-09:
Not really a MightyCore related issue at all, but rather your hardware. I suggest you post your question over at the Arduino forum instead. Lots of people that might help you figure out what's the problem 👍

PiotrC1, 2018-04-13:
I'm using the same USB to serial chip and ATMega 644P and it works with capacitors between 22nF to 1uF (my resistor between Vcc and RST is 4k7). Check your RxD and TxD lines if they are connected correctly. In order to check if lines are OK write short sketch sending something over serial port with `serial.print()` command.
I had the same issue using breadboard but the problem was poor connection of RST capacitor. After soldering wires everything was fixed.

closed 2019-03-18 by MCUdude