MCUdude/MightyCore #92: Addiion of ATMEGA324PB core?

chrismcdaid1, 2018-04-18:
Hi, Just wondering if you have any plans of adding the 324PB variant to the MightyCore.

MCUdude, 2018-04-18:
Hi! I don't have plans adding support for it in the near future, but it shouldn't be too hard. Problem is, I don't have any hardware to test on, and new hardware needs to be fully tested before I can merge it into the master branch.

MCUdude, 2018-04-24:
I've just pushed some [pre-compiled bootloader hex files for ATmega324PB](https://github.com/MCUdude/optiboot_flash/tree/master/atmega324pb) if you'd like to test it.

chrismcdaid1, 2018-04-25:
Cheers,

I have hardware coming this week to test it on.

Regards,
Christopher McDaid
Research Engineer

[wisarlyit2]
T:  +353 (0)74 918 6460
E:  christopher.mcdaid@lyit.ie<mailto:christopher.mcdaid@lyit.ie>

chrismcdaid1, 2018-06-18:
Thats Brilliant MCUdude. I look forward to testing it out. Many thanks.

Luckysin13, 2018-07-11:
Readme still shows ATmega324PB as unsupported on master page.

Thanks for your hard work MCUdude, really appreciate it.

https://github.com/MCUdude/MightyCore#supported-microcontrollers

MCUdude, 2018-07-11:
I'm still working on it. I'll merge it into master when it's ready. Not much work left though