MCUdude/MightyCore #93: PID library compiles with error - missing compiler option -fpermissive

PiotrC1, 2018-04-19:
I'm using standard PID library. Project for ATmega 328 compiles with warnings, while for ATmega 644 with MightyCore compiles with error. Generally, the problem is missing `-fpermissive` option in `compiler.cpp.flags` line in platform.txt file.

There are also some more differencies between Arduino avr standard platforms.txt and Mighty Core one.

per1234, 2018-04-19:
Reference regarding the addition of `-fpermissive` to Arduino AVR Boards:
https://github.com/arduino/ArduinoCore-avr/commit/06868f4cd39a9dba82b5014d1208feb1e1f0d750#diff-d638a47d1ca0e2a5585fb34c7d7fb0f9

The reason for adding this flag was for backwards compatibility because the toolchain update done at that time (Arduino IDE 1.6.10/Arduino AVR Boards 1.6.12) caused some things that had previously been warnings to be upgraded to errors. I believe it also caused some things that had previously been errors to be downgraded to warnings.

MCUdude, 2018-04-20:
What PID library are you using?
Personally I would rather not add the -fpermissive flag. It indicates that there's something wrong in the code somewhere, and that's something I wont "hide" behind a warning. Just remember that warnings are disabled by default in Arduino IDE.

PiotrC1, 2018-04-20:
This is standard PID library form IDE (by Brett Beauregard, version 1.1.1)

MCUdude, 2018-04-20:
I just installed the PID library (by Brett Beauregard). The latest version available was 1.2.0.
All examples compiles just at my computer. I only get a warning when compiling the *PID_RelayOutput* sketch, but no error

PiotrC1, 2018-04-20:
OK, you are right. Version 1.2.0 compiles without any errors.

closed 2018-04-20 by PiotrC1