MCUdude/MightyCore #94: PWM on PD6/PD7?

sgtJohnny, 2018-05-03:
I tried this core on a ATMEGA1284p and wanted to use PWM on PD6 an PD7, but it does not work in Arduino. A digital Write works perfectly fine, so no issue with connected LED. And PWM on PD4/5 works also very fine.

PD6 and PD7 are on "OC2 A /B" When looking at the MightyCore pinout. Is there anything known about my issue, or do i need to activate PWM on these pins with some function?

sgtJohnny, 2018-05-04:
My mega is set to 8Mhz external Crystal

MCUdude, 2018-05-10:
I just connected an 8 MHz crystal and burned the bootloader again. PWM works perfectly fine on D6 and D7 (PB6 and PB7) on my ATmega1284P.

Are you using the latest version of MightyCore? Does it work if you try using the 8MHz internal oscillator? Are you sure you're using the **Standard** pinout?

MCUdude, 2018-05-13:
I need a response from you in order to solve this issue

sgtJohnny, 2018-05-15:
I still have this problem, but i will soon try with a new IC. Yes i'm using standart pinout

MCUdude, 2019-03-18:
I'm closing this because it's not caused by MightyCore. For help with hardware issues, the Arduino forum is a much better place.

closed 2019-03-18 by MCUdude