MCUdude/MightyCore #95: Setting MCUSR to 0 causes boot loop on WDT reset

simonthechipmunk, 2018-05-03:
Hello,

I tried to read the reset cause from the MCUSR register to display an according error message to the user. Unfortunately the MCU keeps resetting itself in an endless loop after it was actually reset by a WDT event.
So far I found that this is caused by a combination of resetting the MCUSR register to 0 (precisely setting the WDRF bit to 0) and the position where the wdt_enable() call is placed in the setup routine.

The MCU used is a Atmega1284p. I have not tested this behavior with other microcontrollers yet.

Here's some example code that reproduces the problem.
My setup takes quite some time because of various initializations so I simulated this by adding a small delay of 200ms:

```
#include <avr/wdt.h>

const int pin = 0;

void setup() {

  //watchdog POSITION 1
  //wdt_enable(WDTO_8S);
  
  pinMode(pin, OUTPUT);
  digitalWrite(pin, HIGH);

  Serial.begin(115200);
  delay(200);

  byte ResetSrc = MCUSR; // save reset source
  MCUSR = 0x00; //cleared for next reset detection
  if (ResetSrc & _BV(PORF)) Serial.println("PORF");
  if (ResetSrc & _BV(EXTRF)) Serial.println("EXTRF");
  if (ResetSrc & _BV(BORF)) Serial.println("BORF");
  if (ResetSrc & _BV(WDRF)) Serial.println("WDRF");

  //watchdog POSITION 2
  wdt_enable(WDTO_8S);
}

void loop() {
  wdt_reset();

  digitalWrite(pin, HIGH);
  delay(200);
  digitalWrite(pin, LOW);
  delay(500);

  if (millis() > 5000)while (true);
}
```
Please note the calls to wdt_enable() at POSITION 1 and POSITION 2:

1: This sketch will work and reset after 8 seconds as it should
2: This will not work. The MCU will run once, reset and then run into a reset loop rapidly flashing the attached LED

Adding calls to wdt_reset() or wdt_disable() to the setup with wdt_enable() at POSITON 2 has no effect on this behavior.

The reason I initialize the watchdog timer at the end of the setup is that I have some network requests running there which can take more than 8 seconds to complete.
Even doing something like the following has no effect on the issue:
```
void setup() {
  //watchdog POSITION 1
  wdt_enable(WDTO_8S);
  wdt_disable();

  ....setup code...

  //watchdog POSITION 2
  wdt_enable(WDTO_8S);
}
```

To me this doesn't make much sense. Am I missing something?

MCUdude, 2018-05-03:
I've not developed the bootloader, so I'm no guru here.
However, have a look at [these source code lines](https://github.com/MCUdude/optiboot_flash/blob/master/optiboot_flash.c#L506-L509). You're actually forcing the bootloader to start on boot when MCUSR is ran.

MCUdude, 2018-05-03:
Seems like this particular "MCUSR fix" only applies for optiboot_flash, which is the version MightyCore uses. You can read the whole discussion here: [https://github.com/Optiboot/optiboot/issues/97](https://github.com/Optiboot/optiboot/issues/97)

simonthechipmunk, 2018-05-04:
I found that discussion too and forgot to link to it. This problem only affects MightyCore 2.0.0 as it's the first one with the MCUSR fix in it. In prior versions it's not even possible to read the contents of MCUSR as it used to be set to 0 by the bootloader. My current workaround was to enable the watchdog at the beginning of setup() and call wdt_reset() when needed until I enter the mainloop.

What bothers me is that the application actually attempts to run. I get parts of my initial serial messages before something resets the MCU for no apparent reason.

After a more thorough look at the optiboot code (and I'm an absolute noob there) I found where the problem might be: https://github.com/MCUdude/optiboot_flash/blob/master/optiboot_flash.c#L888-L892

Optiboot uses a 16ms watchdog to reset the MCU in case of no communication with the bootloader. As you mentioned I'm forcing the bootloader to start so it waits for serial which won't respond of course and then enables the watchdog but it never gets disabled again and kills my application code after 16ms. It also can't detect the successful startup because I'm not setting MCUSR to 0 in that 16ms timeframe.

So the correct approach seems to be setting MCUSR to 0 and also disabling the wdt immediately after application launch. And voilà it is actually working:

```
#include <avr/wdt.h>

const int pin = 0;

void setup() {

  //has to be called immediately
  byte ResetSrc = MCUSR; // save reset source
  MCUSR = 0x00; //cleared for next reset detection
  wdt_disable();
  
  pinMode(pin, OUTPUT);
  digitalWrite(pin, HIGH);

  Serial.begin(115200);
  delay(200);

  if (ResetSrc & _BV(PORF)) Serial.println("PORF");
  if (ResetSrc & _BV(EXTRF)) Serial.println("EXTRF");
  if (ResetSrc & _BV(BORF)) Serial.println("BORF");
  if (ResetSrc & _BV(WDRF)) Serial.println("WDRF");

  //watchdog enable
  wdt_enable(WDTO_8S);
}

void loop() {
  wdt_reset();

  digitalWrite(pin, HIGH);
  delay(200);
  digitalWrite(pin, LOW);
  delay(500);

  if (millis() > 5000)while (true);
}
```

In my mind a call to wdt_disable() alone should have been enough but I guess this is solved for now.
I guess if it's not a bug it's a feature... Should I report this to the optiboot team?

MCUdude, 2018-05-06:
I'm not using the official version of optiboot, but rather a fork. You can [report the issue here](https://github.com/majekw/optiboot).