MCUdude/MightyCore #97: Select Upload Speed from IDE

hallard, 2018-05-14:
Hi there,
I plan to use a 8MHz version of 1284P, I found all I need and really thanks for sharing.
Anyway there are lot of bootloader files and I'm interested by the 250K 8MHz. I already wrote an article some years ago on bootloader ans serial speed [here](https://hallard.me/ulpnode-bootloader-1/) so I know what I need.
But once flashed there is no option to select UART speed (for uploading) on Arduino IDE, this means the need to modify the boards.txt or create a new one because choosing 1284P @ 8MHz always select 115200.
Any option to add upload speed in Menu, would makes sense?
Thanks

hallard, 2018-05-15:
Correct, changing brd file is easy, but on each update it will break my changes.
Anyway, I'll put this on my own custom .brd file. not a problem, I got plenty of custom boards ;-)
Was just asking because having 4 more flash space with 1284P over 328P and flashing at half of the speed of the 328P is totally a non sense from my point of view ;-)

JensGrabner, 2018-05-17:
I use 115200 baud on 12 MHz - Bootloader must compile with U2Xn = 1 ... http://wormfood.net/avrbaudcalc.php?bitrate=115.2k&clock=12&databits=8

MCUdude, 2019-03-18:
For reference you can [read my reply](https://github.com/MCUdude/MiniCore/issues/79#issuecomment-462650620) from a similar question. TL;DR There isn't an elegant way of implementing this anyways.

closed 2019-03-18 by MCUdude