MCUdude/MightyCore #100: [324PB] Compiler does not pick up preproc macros before it's too late

MCUdude, 2018-06-02:
I'm working with the new ATmega324PB (see 324pb_support branch), and it's really annoying that Atmel/Microchip did skip a lot of backwards compatibility #defines in the iom324pb.h file.

Right now I'm trying to get the Wire library to work. I loaded the *i2c_scanner* example sketch, and the compiler threw this error:
```
/Users/Hans/Documents/Arduino/hardware/MightyCore/avr/libraries/Wire/Wire.cpp:85:3: error: 'TWBR' was not declared in this scope
```

The problem is that TWBR is now called TWBR0. In order to fix this I tried to add
```
# define TWBR TWBR0
```
to the *pins_arduino.h* file, but the compiler doesn't pick this up before Wire is compiled.

Does anybody know where I can add these missing #defines so that the compiler picks them up?
I don't want to modify any libraries or the iom324pb.h file; it should within the core itself.

per1234, 2018-06-03:
I know you don't want to modify the library, but it seems like the only solution is to add `#include <pins_arduino.h>` (or more traditionally, `#include <Arduino.h>`) to Wire.h.

MCUdude, 2018-06-03:
Thanks! It turns out that Wire was the only library that didn't include Arduino.h

commit e3a821ee (2018-06-03): Include Arduino.h
closes #100
closed 2018-06-03 by MCUdude