MCUdude/MightyCore #101: Atmega644P Auto Reset

trungdn, 2018-06-05:
Hi,

I'm able to run MightyCore bootloader on Atmega644P with internal 8Mhz clock,  BOD disable, 3V3 but when I try to run code after a period time. It always auto reset the board. 
The design has decoupling between VCC, AVCC, AREF, and GND. 
I don't know what else could lead to auto reset. 
Thank you very much

MCUdude, 2018-06-05:
Do you have a 10k pull up resistor on the RST pin?

trungdn, 2018-06-05:
@MCUdude Yes I do. Interesting, Look like it because of the SD library is not working properly with 8Mhz.
I tried to run with Blink example and it works fine. 
SD.begin(CS_Pin) makes the board reset.
Do you have any idea on how we could fix that? 
Maybe something wrong with SPI speed.
I even tried with 
(!card.init(SPI_HALF_SPEED, chipSelect)) and
(!card.init(SPI_QUARTER_SPEED, chipSelect))

Thanks,

MCUdude, 2018-06-06:
Does it get reset if you run it at 16 MHz?

trungdn, 2018-06-11:
No. It working fine with an external 16Mhz

MCUdude, 2018-06-11:
I'll try to reproduce this issue. Are you able to reproduce this issue when trying the *CardInfo* example?

trungdn, 2018-06-14:
Hi, Look like it is because of the SD libraries. I tried to delete Sd library from IDE and used default SD card from MightyCore. No problem so far. I will try to test with other clocks setting. I haven't looked at the source code but look like something different from the SD(MightyCore)

trungdn, 2018-06-20:
After testing sometimes, the ATmega644P just freeze randomly.
When I run the blink example, it runs well. 
I wonder if Mighty bootloader is incompatible with any other libraries. Since last time I have problems with SD libraries. 
Bellow is the libraries I'm using. Thanks

Using library SD at version 1.0.7 in folder: C:\Users\Trung\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.0\libraries\SD 
Using library RTClib at version 1.2.0 in folder: C:\Program Files (x86)\Arduino\libraries\RTClib 
Using library SHT2x in folder: C:\Program Files (x86)\Arduino\libraries\SHT2x (legacy)
Using library Wire at version 1.0 in folder: C:\Users\Trung\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.0\libraries\Wire 
Using library U8glib in folder: C:\Program Files (x86)\Arduino\libraries\U8glib (legacy)
Using library Low-Power at version 1.6 in folder: C:\Users\Trung\Documents\Arduino\libraries\Low-Power 
Using library SPI at version 1.0 in folder: C:\Users\Trung\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.0\libraries\SPI

MCUdude, 2018-06-20:
The bootloader should not affect the running program. It's physically separated from the program (in the flash memory. If you're experiencing random resets/hangs you should really check your code. You may be running out of RAM.

trungdn, 2018-06-20:
Thank you very much. Even when I run an U8glib example but it seems not working. It just freezes the screen now. It's weird, since last time I able to run it continuous few days without problems but it happened recently. Nothing change in the code. The RAM only uses 10%. Free 3648 bytes. 
Actually, I run my program few months on Sanguino bootloader without problems but seems they could not go to deep sleep mode with BOD off. That I found Mighty much better. 

This is my setup when bootloader.
Board: "ATmega644"
Variant: "644P/644PA"
BOD: "Disabled"
Pinout: "Standard"
Clock: "16 Mhz external"
Compiler LTO: "Disabled (default)"
Port: ""COM35"
Programmer: "Arduino as ISP (MightyCore)"

I followed this tutorial to burn bootloader 
https://www.arduino.cc/en/Tutorial/ArduinoToBreadboard

Thank you very much for your quick reply.

SpenceKonde, 2020-05-24:
Are you doing this on a breadboard? Do you have the 0.1uF ceramic capacitors between each pair of power and ground pins, located as close to the chip as possible? (the length of the electrical path should be an inch *maximum*. Omitting these necessary components can result in unpredictable and hard to debug stability problems. (as that stupid tutorial does - they have been told about it countless times and choose not to fix it; it's as if the Arduino people have a vested interest in people not making reliable projects on breadboard or something). And some of the tutorials that DO show these "decoupling caps" put them inches away on the breadboard - since their purpose is to counteract the inductance of the several inches of wire from the power supply and ensure that there isn't a glitch on the power supply that the device "sees" - now how are they going to do that with SEVERAL INCHES OF WIRE between them and the chip?! But yeah - without decoupling caps, you're operating the part outside of it's design parameters and are in [nasal demon](https://en.wikipedia.org/wiki/Undefined_behavior) territory. 

Sometimes libraries use "dynamic allocation", which does not show up in the ("find in files" search from your favorite programming text editor on the library folder for `malloc(` - if you find it, some/much of the memory that the library uses will not be counted in the ram usage (because it's allocated at runtime, not compile time)

Also, it's important to determine if it's your code crashing itself, or your code crashing the controller on the screen. If it's hanging (as opposed to resetting), it's more likely getting stuck in some busy-wait loop, waiting for something to happen, but that something never happens. Put a debug print statement, or obvious light flashing pattern in at the start of setup (as soon as you set the debug LED pin OUTPUT or setup Serial) so you can make sure it's not resetting (which would mean it is a straight-up defect in your code - assuming you aren't missing decoupling caps) 

Actually - the fact that you wanted to run without BOD enabled - I don't suppose that was because you had problems with the BOD triggering when it shouldn't? If you did see that before, and you don't think you should have been, turning off the BOD just hides the actual problem, which would likely be missing decoupling capacitors. 

Anyway, why am I here doing his tech support for him?

MCUdude, 2020-09-06:
Haven't heard anything for a long time, so I'm closing this. Haven't heard of other users experiencing the same issue either

closed 2020-09-06 by MCUdude