MCUdude/MightyCore #108: Noob question: Does it support "normal arduino code"

hugopilot, 2018-07-11:
To make it a bit shorter: 

Your provided C libraries look more low-level than I code in. You know, standard Arduino C++.  But does this plugin support that as well? I really want to use the ATMEGA644 but if I could program it as if I am programming a UNO, without low-level AVR code, that would be awesome!

(P.S I don't know where to put this question so I decided to put it here)

MCUdude, 2018-07-11:
Hi! Yes, MightyCore supports the standard "Arduino language" (or more correct; API). The code you're already written for the uno will most likely work with the ATmega644 without any modifications!

closed 2018-07-11 by MCUdude

hugopilot, 2018-07-11:
Good to hear! Thanks for the fast replay