MCUdude/MightyCore #112: Using Analog pins of ATmega32 as Digital I/O

Hamimn, 2018-08-06:
Hi,

I've tried to right a HIGH value on a an analog pin of ATmega32. For ADC0 of ATmega32, I've wrote:
`pinMode(24,OUTPUT);`
`digitalWrite(24,HIGH);`
But it doesn't work. Isn't it possible to define an analog pin as a digital I/O in MightyCore?

MCUdude, 2018-08-06:
> Isn't it possible to define an analog pin as a digital I/O in MightyCore?

All pins on the ATmega32 are "digital" by default. Are you sure you're measuring  the correct pin (physical pin 40).

Hamimn, 2018-08-06:
Thanks. Yes, I've tried multiple analog pins but I've got the same result. May be the pins are not working.

MCUdude, 2018-08-06:
Please post the entire code. Did you upload using a USB to serial adapter or an ISP programmer?

Hamimn, 2018-08-06:
I used a USB to serial adapter to upload the code.

`int pin =  24;
void setup() {
  pinMode(pin, OUTPUT);
}

void loop() {
  digitalWrite(pin, HIGH);   
  delay(1000);                       
  digitalWrite(pin, LOW);    
  delay(1000);                     
}`

MCUdude, 2018-08-06:
Well, then your chip must be defective. I just uploaded your code to an ATmega32 board I have laying around, and pin 24 works perfectly..

closed 2018-08-07 by Hamimn