MCUdude/MightyCore #113: I can't burn bootloader to a Atmega1284p with Arduino as ISP

gusantor, 2018-08-12:
Hello, thank you for this work

I can't burn bootloader to a Atmega1284p with Arduino as ISP (I've successfully burn bootloader on many atmega328)

I just try to reproduce burn bootloader for the Atmega328, on a 1284p, connecting:

UNO                  1284p
pin10    -->         reset, pin9
pin11    -->         MOSI, pin5
pin12    -->         MISO, pin6
pin13    -->         SCK, pin7
GND     -->         GND
Vcc       -->         Vcc

but receive error, full response I receive is as follow

	Arduino:1.8.0 (Linux), Tarjeta:"ATmega1284, Standard, 1284P, 2.7v, Disabled (default), 16 MHz external"

	/home/mauricio/.arduino15/packages/arduino/tools/avrdude/6.3.0-arduino9/bin/avrdude -C/home/mauricio/.arduino15/packages/MightyCore/hardware/avr/2.0.1/avrdude.conf -v -patmega1284p -cstk500v1 -P/dev/ttyACM0 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xfd:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xf7:m 

	avrdude: Version 6.3, compiled on Jan 17 2017 at 06:01:25
			     Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
			     Copyright (c) 2007-2014 Joerg Wunsch

			     System wide configuration file is "/home/mauricio/.arduino15/packages/MightyCore/hardware/avr/2.0.1/avrdude.conf"
			     User configuration file is "/home/mauricio/.avrduderc"
			     User configuration file does not exist or is not a regular file, skipping

			     Using Port                    : /dev/ttyACM0
			     Using Programmer              : stk500v1
			     Overriding Baud Rate          : 19200
			     AVR Part                      : ATmega1284P
			     Chip Erase delay              : 55000 us
			     PAGEL                         : PD7
			     BS2                           : PA0
			     RESET disposition             : dedicated
			     RETRY pulse                   : SCK
			     serial program mode           : yes
			     parallel program mode         : yes
			     Timeout                       : 200
			     StabDelay                     : 100
			     CmdexeDelay                   : 25
			     SyncLoops                     : 32
			     ByteDelay                     : 0
			     PollIndex                     : 3
			     PollValue                     : 0x53
			     Memory Detail                 :

			                              Block Poll               Page                       Polled
			       Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
			       ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
			       eeprom        65    10   128    0 no       4096    8      0  9000  9000 0xff 0xff
			       flash         65    10   256    0 yes    131072  256    512  4500  4500 0xff 0xff
			       lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
			       lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
			       hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
			       efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
			       signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
			       calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

			     Programmer Type : STK500
			     Description     : Atmel STK500 Version 1.x firmware
			     Hardware Version: 2
			     Firmware Version: 1.18
			     Topcard         : Unknown
			     Vtarget         : 0.0 V
			     Varef           : 0.0 V
			     Oscillator      : Off
			     SCK period      : 0.1 us

	avrdude: AVR device initialized and ready to accept instructions

	Reading | ################################################## | 100% 0.02s

	avrdude: Device signature = 0xff0000
	avrdude: Expected signature for ATmega1284P is 1E 97 05
			     Double check chip, or use -F to override this check.

	avrdude done.  Thank you.

	Error quemando bootloader

Please any advice is highly appreciated

Thanks

gusantor, 2018-08-12:
Sorry me, my foult, bad pin connection :(

Now it works !

Thanks

closed 2018-08-12 by gusantor

MCUdude, 2018-08-13:
Great!