MCUdude/MightyCore #115: Not able to upload code to Atmega16

manasdas17, 2018-08-27:
Hi, I am using an **ATmega16 with external 16MHz crystal oscillator and using an stk500v2 programmer. But getting this issue. PFA for screenshot**. But i am able to upload  the .hex files generated by Arduino by stk500v2 from ubntu terminal by "avrdude -p atmega16 -P /dev/ttyACM0 -c stk500v2 -v -U flash:w:blink_1tmega16.ino.hex".What is the issue?
[image]

prampec, 2019-02-22:
Similar problem here with ATmega1284. I have tried to set up things in a breadboard. Burn bootloader was successful. I can upload using programmer, but cannot upload from serial: not in sync: resp=<random>
Serial communication works on 19200, and serial upload resets the MCU as expected.
Using FT232RL serial module on 5V, 10K pullup on reset, 100nF between reset and DTR.
Is there anything I can try out?
[image]

martin323232, 2019-09-07:
I have the same problem with 1284P. It works fine with Arduino 1.0.5.
Same setup as prampec, except I used 1.05 to bootloaded it.

mrguen, 2019-10-08:
Hi martin323232,

Did you solve the issue? What do you mean by Arduino 1.0.5 : the IDE (this old?)  or board definitions?

prampec, 2019-10-10:
My issue was solved at [#136 ](url)

mrguen, 2019-10-10:
Thanks. The link is dead...

prampec, 2019-10-10:
(fixed the link above)

MCUdude, 2020-03-15:
@manasdas17 The issue you're having is because you're trying to upload code using a USB to serial adapter in Arduino IDE. Select your programmer in the Programmers menu and hold shift down while pressing Upload.

closed 2020-03-15 by MCUdude