MCUdude/MightyCore #117: 644 PU 20mhz

geppoleppo, 2018-09-02:
Hi,
i try to write bootloader on atmega 644-20-PU but the Arduino IDE respond with the error: Expected signature for ATmega644P is 1E 96 0A.
Any Ideas?
the detailed error is: 
Arduino:1.8.5 (Windows 10), Scheda:"ATmega644, Standard, 644P / 644PA, 2.7v, Disabled (default), 20 MHz external"

C:\Program Files (x86)\Arduino\hardware\tools\avr/bin/avrdude -CC:\Users\Debora\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.1/avrdude.conf -v -patmega644p -cstk500v1 -PCOM3 -b19200 -e -Ulock:w:0x3f:m -Uefuse:w:0xfb:m -Uhfuse:w:0xd6:m -Ulfuse:w:0xf7:m 

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\Debora\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.1/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega644P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       2048    8      0  9000  9000 0xff 0xff
           flash         33     6   256    0 yes     65536  256    256  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000 (retrying)

Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000 (retrying)

Errore durante la scrittura del bootloader
Reading | ################################################## | 100% 0.02s

avrdude: Device signature = 0x000000
avrdude: Yikes!  Invalid device signature.
         Double check connections and try again, or use -F to override
         this check.

avrdude done.  Thank you.

MCUdude, 2018-09-03:
A device signature of 0x000000 indicates no communication with the microcontroller. Check the connections.

geppoleppo, 2018-09-03:
ok, hi have retry...
the error is:
Arduino:1.8.5 (Windows 10), Scheda:"ATmega644, Standard, 644P / 644PA, 2.7v, Disabled (default), 16 MHz external"

avrdude: Version 6.3, compiled on Jan 17 2017 at 12:00:53
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2014 Joerg Wunsch

         System wide configuration file is "C:\Users\silvano\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.1/avrdude.conf"

         Using Port                    : COM3
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
         AVR Part                      : ATmega644P
         Chip Erase delay              : 55000 us
         PAGEL                         : PD7
         BS2                           : PA0
         RESET disposition             : dedicated
         RETRY pulse                   : SCK
         serial program mode           : yes
         parallel program mode         : yes
         Timeout                       : 200
         StabDelay                     : 100
         CmdexeDelay                   : 25
         SyncLoops                     : 32
         ByteDelay                     : 0
         PollIndex                     : 3
         PollValue                     : 0x53
         Memory Detail                 :

                                  Block Poll               Page                       Polled
           Memory Type Mode Delay Size  Indx Paged  Size   Size #Pages MinW  MaxW   ReadBack
           ----------- ---- ----- ----- ---- ------ ------ ---- ------ ----- ----- ---------
           eeprom        65    10   128    0 no       2048    8      0  9000  9000 0xff 0xff
           flash         33     6   256    0 yes     65536  256    256  4500  4500 0xff 0xff
           lock           0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           lfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           hfuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           efuse          0     0     0    0 no          1    0      0  9000  9000 0x00 0x00
           signature      0     0     0    0 no          3    0      0     0     0 0x00 0x00
           calibration    0     0     0    0 no          1    0      0     0     0 0x00 0x00

         Programmer Type : STK500
         Description     : Atmel STK500 Version 1.x firmware
         Hardware Version: 2
         Firmware Version: 1.18
         Topcard         : Unknown
         Vtarget         : 0.0 V
         Varef           : 0.0 V
         Oscillator      : Off
         SCK period      : 0.1 us

Errore durante la scrittura del bootloader
avrdude: AVR device initialized and ready to accept instructions

Reading | ################################################## | 100% 0.03s

avrdude: Device signature = 0x1e9609 (probably m644)
avrdude: Expected signature for ATmega644P is 1E 96 0A
         Double check chip, or use -F to override this check.

avrdude done.  Thank you.

MCUdude, 2018-09-03:
As you can can see you're now able to communicate with the device, and it turns out that you have an ATmega644/A, not an ATmega644P/PA. you can change this in the variants menu.

MCUdude, 2018-09-04:
if you read the error carefully you can see that you're no longer able to communicate with the microcontroller. Again, it's a hardware issue, and not software related.

I suggest you ask over at the Arduino or AVRfreaks forum instead, as this issue has nothing to do with MightyCore, but rather your setup.

closed 2018-09-04 by MCUdude