MCUdude/MightyCore #121: How to change fuses for Atmega16A;Atmega32A.

gavron04, 2018-10-04:
Hello. I need to change original fuses from boards.txt (C:\Users\X\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\2.0.1\) to:

Atmega16;32:

8-16MHz external crystal for hfuse - C9; lfuse - FF

I need to change 
32.menu.clock.8MHz_external.bootloader.high_fuses=0xc6 >0xc9
32.menu.clock.16MHz_external.bootloader.high_fuses=0xc6 > 0xc9
16.menu.clock.16MHz_external.bootloader.high_fuses=0xc4 >0xc9 16.menu.clock.8MHz_external.bootloader.high_fuses=0xc4 > 0xc9

but where I can find low_fuses? 

Additionally. Its possible to generate only hex file without fuses inside? I have wrote some code on Arduino for Atmega16;32 , then I press export compileted program (hex file) and then Im programming Atmega by USBasp. Fuses Im changing in avrdude.

Regards

MCUdude, 2018-10-04:
Hi!

The low fuse on the ATmega8535, ATmega16 and ATmega32 is modified in a rather clever way to achieve what I want.
The physical low fuse value is located [here](https://github.com/MCUdude/MightyCore/blob/8c4e738d1bfdab9c782c21e2fc31135a51f84fde/avr/boards.txt#L627), and can be changed to what ever you like. Just make sure to remove **0b{bootloader.bod_bits}{bootloader.sut_cksel_bits}**. The reason why it's done this way (low fuse represented in binary with *bod_bits* and *sut_cksel_bits) is because this is the only way to manipulate a single bit in the boards.txt file use a fuse calculator ([Engbedded for instance](http://engbedded.com/fusecalc))* if you want to check what the bits represents.

Hex files generated for AVRs does not contain any fuse bits. These needs to be sat separately using Avedude.

gavron04, 2018-10-04:
Hi! Thanks for answer.

Are You sure that hex file didnt contains fuse bits? Few times after programming only hex file the fuse bits was changed itself.

So if fuse bits I need change separately using Avrdude for what I need to apply BOD level; Clock; Compiler LTO in arduino settings?

MCUdude, 2018-10-04:
> Are You sure that hex file didnt contains fuse bits? Few times after programming only hex file the fuse bits was changed itself.

100% sure. If you turn on verbose upload in Arduino IDE, you'll see that when burning bootloader it first sets the fuse bits and then write the bootloader hex. The compiler LTO has nothing to do with the fuse bits. That's only to tell the compiler to turn LTO on or off.

closed 2018-12-06 by MCUdude