MCUdude/MightyCore #122: Strange delay when starting serial

Kabron287, 2018-10-08:
I do understand that is no bootloader issue, but I cannot figure out how it could be.
I wrote tons of code, but facing this for the first time.
Simple test code:
`int main(void)
{       
Serial.begin(115200);
Serial.println("Starting...");
pinMode(LED_BUILTIN, OUTPUT);
  while(1)
  {	  
    digitalWrite(LED_BUILTIN, HIGH);
    _delay_ms(1000);
    Serial.println("Hello");
    digitalWrite(LED_BUILTIN, LOW);
    _delay_ms(1000);
}
`

When code starting I see in the serial terminal:
**St**    [8s pause]   **rting** [1s pause] **. . .**    // meanwhile LED flashing properly during these pauses
**Hello
Hello**
etc.

uC Atmega324pa, XTAL 12MHz, Arduino IDE 1.8.5, VisualMicro 1808.30.4
Tested on two different PC Windows 10 x64
Did anybody met such magic?

Kabron287, 2018-10-08:
Well, converting the code to standard arduino form:
void setup(){}
void loop(){}
solved the problem.
So, the question is why all examples are decorated in such strange notation?

MCUdude, 2018-10-08:
I'll have to look closer at this later, but try to replace _delay_ms() with delay() to see if it helps.

MCUdude, 2018-12-06:
Did the behavior change when using `delay` instead of `_delay_ms`?

MCUdude, 2018-12-17:
I think I figured it out. The Serial library won't work properly if global interrupts isn't enabled (`sei()`).
By using setup() and loop() you're actually running the [init()](https://github.com/MCUdude/MightyCore/blob/1856854fb82703c5bc283dac7b53c1aa5e383b9d/avr/cores/MCUdude_corefiles/wiring.c#L241) function, which again enables global interrupts.

Add `sei();` to your initialization part and it should be fine.

closed 2018-12-17 by MCUdude