MCUdude/MightyCore #123: How to define bobuino pinout variant in platformio.ini?

netbomo, 2018-10-28:
Hello,
I'm trying to compile code under Atom with the PlatformIO package. 

I've set my platformio.ini : 
```
[env:mightycore1284]
platform = atmelavr
board = mightycore1284

; change microcontroller
board_build.mcu = atmega1284p

; change MCU frequency
board_build.f_cpu = 16000000L

framework = arduino
lib_deps =
   ...
```

Unfortunately, I've not found in the PlatformIO documentation how to set pinout/ board variant. Do you know how to do that?
That can be help full for other to add this on your  README.md
Thanks,
gilou

chadwpalm, 2018-12-27:
From what I can tell, PlatformIO only supplies the standard pinout variant for MightyCore ATmega1284. I had to navigate to the directory that contains the pinout file and replace it with the bobduino pinout file. There may be an easier way, but so far that's the only way I know of. I'm not sure which platform you are on, but in Linux it's in: /home/(user)/.platformio/packages/framework-arduinoavr/variants/mightycore
Make sure you backup the file if you want to retain the standard pinout or else you'll have to redownload from here.

MCUdude, 2019-08-02:
There's been some changes to the MightyCore PlatformIO integration lately. Everything you need to know, including how to use the Bobuino pinout can be found here https://github.com/MCUdude/MightyCore/blob/master/PlatformIO.md

closed 2019-08-02 by MCUdude